raulmur/ORB_SLAM2 #1: Runtime stability

genawass, 2016-01-25:
Hello
First of all, thank you for sharing OrbSlam library.
Second, after checking it with TUM database it appears that the algorithm is not stable:

1. Several initializations with the same database produce different results.
2. Sometimes SLAM stuck in reset mode (Resetting database) for almost all datasets from freiburg1. 
3. Sometimes track is lost and SLAM exits with error.

I've noticed one smooth run on "rgbd_dataset_freiburg2_360_hemisphere" while also some initializations appear to stuck or loose track.
Also it seems to work rather slowly (5-10 fps) sometimes slower.

raulmur, 2016-01-25:
-Could you please provide more information? 
-Are you testing Monocular or RGB-D.?
-In which datasets does it fail to initialize? Take into account that some sequences of the TUM dataset (especially the freiburg1 sequences) are not suitable for monocular cameras.

The system requieres a powerful computer (e.g. i7) to work properly.

genawass, 2016-01-25:
Thank you for your reply.
I was testing Monocular Slam.
Datasets: rgbd_dataset_freiburg1_360, rgbd_dataset_freiburg1_room appear to stuck at "Resetting database" mode.
rgbd_dataset_freiburg3_walking_halfsphere also appears to stuck sometimes, sometimes it is working.