raulmur/ORB_SLAM2 #3: ROS Integration

hbadino, 2016-01-26:
Hi Raul,

the current version doesn't seem to be integrated with ROS. Are you planning to do it anytime soon?

Hernan

weblucas, 2016-01-26:
use the 	plnegre fork. It is already catkinzated. Now the ros version is inside the Examples

raulmur, 2016-01-27:
As @weblucas said you can find a ROS package at Examples/ROS. It has two nodes, one for monocular and another for RGB-D. I have tested it with Indigo and Hydro. I will write the node for stereo soon (I want to test it with a camera in hand and I did not have the chance yet). Anyway I saw plnegre fork also includes a node for stereo.

plnegre, 2016-01-27:
Hello! In the fork https://github.com/plnegre/ORB_SLAM2, I've catkinized the ROS nodes. Also, I've prepared and tested the ROS stereo node using a bublebee 2. It works perfectly.

Thank you @raulmur because ORB_SLAM2 is one of the best plug'n'play SLAM solutions that works really well on several environments ;)

weblucas, 2016-01-27:
Fully agreed. Really thanks for sharing. I can see your citations sky
rocketing in the near future ;)

raulmur, 2016-01-28:
Thanks guys!. @plnegre I am very happy to see you successfully tried with the bumblebee2.