raulmur/ORB_SLAM2 #4: Pangolin X11: Unable to retrieve framebuffer options

sunstarchan, 2016-01-28:
Hello,

when I follow these steps to run KITTI monocular example, I get following error.

terminate called after throwing an instance of 'std::runtime_error'
what(): Pangolin X11: Unable to retrieve framebuffer options
Aborted (core dumped)

I'm running it with Ubuntu running in Parallels on Mac.
What's the main reason for this issue?

Thanks.

raulmur, 2016-01-28:
When do you get this error?

sunstarchan, 2016-01-29:
Thanks for your comment.

when I just run the KITTI example, after loading vocabulary.
I look into the Pangolin code, it seems that glXChooseFBConfig returned a NULL pointer in CreateX11Window(). 

I'm just wondering whether or not my virtual Ubuntu system doesn't connect to a real display, as I tested all Pangolin examples on Mac, all of them can work correct. 

So does Pangolin support virtual system running in Parallels, VirtualBox or VMware player?

sunstarchan, 2016-01-29:
I got another notebook with Ubuntu install on real hardware, and followed all these steps, I could run these examples. So maybe Pangolin needs to a real display. 

BTW, your work is awesome, thanks for sharing.

sunstarchan, 2016-02-02:
I finally got ORB_SLAM2 run successfully on my virtual Ubuntu system by commenting two lines of Pangolin as explained here. 

https://github.com/stevenlovegrove/Pangolin/issues/74#issuecomment-178248684

closed 2016-02-02 by sunstarchan

sakurazhu, 2016-05-27:
i have met the same problem.I have tired the  method above ,but  it does not work for me .
my device is ubuntu 14.04   on  vmware   workstation 10.0.4