raulmur/ORB_SLAM2 #8: Unlimited parallel compilation

tomas789, 2016-02-01:
I had a problem with parallel compilation. At the end of `build.sh` file is command 
```
make -j
```
That causes a lot of processes to fire up. When I'm compiling on computer with low memory (actually 2GB), then it runs out of memory. I propose add some limit to parallel compilation like 
```
make -j 4
```
which should be enough because ORB-SLAM doesn't take that long to compile

MarcGyongyosi, 2016-02-01:
this is common when you compile on a computer with less memory, I don't think there's a need to change this, since @raulmur clearly recommends this be run on a high end platform

Nican, 2016-02-06:
Might I recommend using "make -j $(nproc)", such that it uses one process per core for any machine.