raulmur/ORB_SLAM2 #10: Save and load map

MarcGyongyosi, 2016-02-02:
Hey @raulmur great work with this update to ORB! I would love to know what the status is on the save/load map functionality. If you give me some directions I can also offer to implement this quickly.

Are there any specific challenges that need to be overcome to implement this? 

Thanks!

Marc

raulmur, 2016-02-02:
I do not plan to do it in the short-term. It should be straightforward, still the challenge is in doing it efficiently. So if you want to have that functionality soon I encourage you to do it yourself. Basically you need to store the keyframe database for place recognition, and all MapPoints and KeyFrames in the map.

MarcGyongyosi, 2016-02-02:
ok thanks! I will report back once I have time to implement it....

closed 2016-02-02 by MarcGyongyosi