raulmur/ORB_SLAM2 #9: Creating my own vocabulary

alejodosr, 2016-02-02:
Hi, 

I am trying to re-create my own vocabulary file, in order to adapt the ORB-SLAM2 algorithm to the camera streaming in a Gazebo simulation. I use DBow2 to create both the vocabulary file _*_voc.yml.tgz_ and the _*_db.yml.tgz_. 

However, the ORB-SLAM2 algorithm requires a _*.txt_ vocabulary file (that is suppose to be ¿serialized?).

¿How can I generate the *.txt from the files I have mentioned?

Thanks

raulmur, 2016-02-02:
Use saveToTextFile function in TemplatedVocabulary.h (the one I provide, not the original one).

closed 2016-02-02 by raulmur

alejodosr, 2016-02-02:
Thanks, I'll try that funciton. To properly generate the vocabulary, is it needed to use ORB-features? (or SURF can be used..) Sorry if it is a silly question, I am a beginner in this field.

raulmur, 2016-02-02:
You have to use ORB features as long as you continue using them in ORB-SLAM.

agharsh11, 2017-07-14:
@raulmur i am trying to implement ORB SLAM for underwater images should i make a new vocabulary or the one provided is sufficient....
thanks

manthan99, 2020-07-16:
Have a look at this repo for step by step procedure to create your own vocabulary:
https://github.com/manthan99/ORB_SLAM_vocab-build