raulmur/ORB_SLAM2 #12: Keyframe trajectory explanation

Markbanfd, 2016-02-03:
Hello,

Could you give some further insight about the output keyframe trajectory for the monocular examples?
The first column is the timing. Then is t (the next 3 columns) the camera position in (x,y,z) and q (the last four columns) the camera orientation in quaternion? 

Does this:
 cv::Mat R = pKF->GetRotation().t();
return the orientation in (x,y,z) (no quaternion)? 

Thanks.

raulmur, 2016-02-03:
SaveKeyFrameTrajectoryTUM() save keyframe poses in the format of the TUM RGB-D dataset. Check http://vision.in.tum.de/data/datasets/rgbd-dataset

Markbanfd, 2016-02-04:
Thanks. Could you point me which parts give the the 3x1 translation vector and 3x3 rotation matrix values?

closed 2016-02-05 by Markbanfd