raulmur/ORB_SLAM2 #13: Building ORB-SLAM2 library and TUM/KITTI examples

marcelinomalmeidan, 2016-02-03:
Hi! I was following the tutorial for building ORB-SLAM2 and when I got to the point of ./build.sh, I found a few problems that (seemingly) I was able to solve, but others I wasn't.
- First, there would be a problem in file ORBextractor.cc, which required to add the header:
#include <opencv2/opencv.hpp> (solution taken from https://github.com/raulmur/ORB_SLAM/issues/44)
- Next, I had similar problems in system.cc, stereo_kitti.cc and mono.cc. I solved these errors by adding the header to those files (these files were not able to execute the function setprecision()):
#include <iomanip>

If I didn't do anything that should not be done, I guess it was supposed to work. However, I get the following output when hitting ./build.sh (I get the libORB_SLAM.so output, but not the executables for mono_tum, mono_kitti, rgbd_tum, stereo_kitti):

Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mma2739/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mma2739/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mma2739/ORB_SLAM2/build
[ 82%] Built target ORB_SLAM2
Scanning dependencies of target mono_kitti
Scanning dependencies of target stereo_kitti
Linking CXX executable ../Examples/RGB-D/rgbd_tum
Linking CXX executable ../Examples/Monocular/mono_tum
[ 86%] [ 91%] Building CXX object CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o
Building CXX object CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o
/usr/bin/ld: warning: libopencv_core.so.3.1, needed by ../Thirdparty/DBoW2/lib/libDBoW2.so, may conflict with libopencv_core.so.2.4
/usr/bin/ld: CMakeFiles/rgbd_tum.dir/Examples/RGB-D/rgbd_tum.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/usr/local/lib/libopencv_core.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/RGB-D/rgbd_tum] Error 1
make[1]: *** [CMakeFiles/rgbd_tum.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/bin/ld: warning: libopencv_core.so.3.1, needed by ../Thirdparty/DBoW2/lib/libDBoW2.so, may conflict with libopencv_core.so.2.4
/usr/bin/ld: CMakeFiles/mono_tum.dir/Examples/Monocular/mono_tum.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/usr/local/lib/libopencv_core.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/Monocular/mono_tum] Error 1
make[1]: *** [CMakeFiles/mono_tum.dir/all] Error 2
Linking CXX executable ../Examples/Monocular/mono_kitti
Linking CXX executable ../Examples/Stereo/stereo_kitti
/usr/bin/ld: warning: libopencv_core.so.3.1, needed by ../Thirdparty/DBoW2/lib/libDBoW2.so, may conflict with libopencv_core.so.2.4
/usr/bin/ld: CMakeFiles/mono_kitti.dir/Examples/Monocular/mono_kitti.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/usr/local/lib/libopencv_core.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/Monocular/mono_kitti] Error 1
make[1]: *** [CMakeFiles/mono_kitti.dir/all] Error 2
/usr/bin/ld: warning: libopencv_core.so.3.1, needed by ../Thirdparty/DBoW2/lib/libDBoW2.so, may conflict with libopencv_core.so.2.4
/usr/bin/ld: CMakeFiles/stereo_kitti.dir/Examples/Stereo/stereo_kitti.cc.o: undefined reference to symbol '_ZN2cv6String10deallocateEv'
/usr/local/lib/libopencv_core.so.3.1: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [../Examples/Stereo/stereo_kitti] Error 1
make[1]: *** [CMakeFiles/stereo_kitti.dir/all] Error 2
make: *** [all] Error 2

Beewe, 2016-02-04:
I am not sure but the output says you're linking DBoW2 with opencv3.1 library and the Examples with opencv2.4. I would first try to build and link all of them with the same version.

marcelinomalmeidan, 2016-02-04:
Thank you  @Beewe! It worked!
I uninstalled opencv, and reinstalled 2.4.11! Works as a charm ;-)

closed 2016-02-04 by marcelinomalmeidan

vinaykaushik15, 2016-09-02:
Instead of installing opencv again, just edit the cmakelists.txt (all) and remove version number written i.e. 2.4 from the OPENCV REQUIRED line.