raulmur/ORB_SLAM2 #17: Map Viewer Top View

dk683, 2016-02-05:
Hello, for the Map Viewer, would there be a way to set the view as Top View, like this video (https://www.youtube.com/watch?feature=player_embedded&v=LnbAI-o7YHk)?
Thank you for your help!

marcelinomalmeidan, 2016-02-05:
I change the view using the three buttons of my mouse. Ends up being intuitive to use! ;-)

marcelinomalmeidan, 2016-02-05:
Also, I guess you can change the viewer parameters in the .yaml file you are using (the following is in the TUM1.yaml):
Viewer.ViewpointX: 0
Viewer.ViewpointY: -0.7
Viewer.ViewpointZ: -1.8
Viewer.ViewpointF: 500

dk683, 2016-02-08:
Hello @marcelinomalmeidan! Modifying the Viewer.Viewpoint* parameters in *.yaml did the trick!
Thank you for your help!
:+1: :+1:

closed 2016-02-08 by dk683

LinHuican, 2016-03-09:
dd

LinHuican, 2016-03-09:
@marcelinomalmeidan @dk683 Current Frame work well, while the map viewer cannot, any advices?

Thank you for your help!

dk683, 2016-03-09:
Hello @LinHuican. I am sorry. I could not understand your question.
Could you be more specific?! :-)

LinHuican, 2016-03-09:
Hi, Thank you for you attention.@dk683. I run monocular examples on ubuntu 12.04, and the  "Current Frame" window work as expected, while the "Map Viewer" Window just black, not working.
Thank you for your help!

dk683, 2016-03-09:
Hmm ... that is strange. Could you double check you installed everything (including prerequisites) correctly?

Additionally, are you running the TUM dataset or KITTI dataset example?
Also, are you running the ROS  monocular version?

LinHuican, 2016-03-10:
Yes, so strange. I cannot install libtiff5-dev, which is optional dependencies for video input of Pangolin, I install libtiff4 instead. (When I try to remove libtiff4 to install libtiff5-dev, I broke the ubuntu. )
I am running the TUM dataset example, without ROS.

Now, I install ubuntu 14.04.3, and the problem is missing.

Thank you very much!!!

dk683, 2016-03-11:
I am glad to hear! :D

zzzzzq, 2016-07-11:
@dk683 @LinHuican Hello！Sorry to bother you, but I met the same question as @LinHuican ,Current frame works well but Map Viewer is all black. I tried to solve this for days but I failed....I have installed all dependencies and I use ROS indigo with xtion depth camera.

LinHuican, 2016-07-12:
Hello @zzzzzq, your ubuntu version is 12.04? Well, when I installed ubuntu 14.04, the problem was missing. So I am sorry to say I cannot help, and if you can solve the problem, could you please share it?

zzzzzq, 2016-07-12:
@LinHuican Thank you! But my version is 14.04......I have reinstalled it sever times but it doesn't work...maybe I should consider some other slam software

zhw5772, 2017-07-21:
@zzzzzq  Have you solved the problem? I met the same problem.
14.04,the map viewer is black.
If you can solve is, could you please share it?

zzzzzq, 2017-07-21:
@zhw5772 I vaguely remember it was because I didn't install Pangolin correctly.

zhw5772, 2017-07-21:
@zzzzzq  Thank you!
I have re-installed Pangolin for many times.
The map viewer still did not work.
I am going crazy...

zzzzzq, 2017-07-21:
@zhw5772 May be you should try to delete build/ file and recompile it or you can find another way to test if your Pangolin is installed correctly