raulmur/ORB_SLAM2 #22: Pangolin error

chaiein, 2016-02-15:
make[2]: *** No rule to make target `../Pangolin/build/src/libpangolin.so', needed by `../lib/libORB_SLAM2.so'.  Stop
How to resolve this error?

chaiein, 2016-02-16:
make[2]: *** No rule to make target `../Pangolin/build/src/libpangolin.so', needed by `../lib/libORB_SLAM2.so'.  Stop.
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2

tdnet12434, 2016-02-28:
try to edit CMakeList.txt

from
find_package(Pangolin REQUIRED)
to
find_package(Pangolin 0.2 REQUIRED)

chaiein, 2016-02-28:
I tried your suggestion but still the error is not cleared. Its showing the same error.

amiltonwong, 2016-03-29:
Did u run command:
cmake -DCPP11_NO_BOOST=1 ..

And what is the output? Any error appears?

chaiein, 2016-03-29:
:~/ORB_SLAM2$ cmake -DCPP11_NO_BOOST=1
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is GNU 4.8.4
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
Build type: Release
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using flag -std=c++11.
-- Found Eigen3: /usr/include/eigen3 (Required is at least version "3.2.0") 
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CPP11_NO_BOOST

-- Build files have been written to: .home/chaiein/ORB_SLAM2

chaiein, 2016-03-29:
Thank you for the solution. I used ./build.sh 
Then cmake -DCPP11_NO_BOOST=1 .. its successfully built

chaiein, 2016-03-29:
When used the same command under Pangolin package ..

I get the following error
Building CXX object src/CMakeFiles/pangolin.dir/video/drivers/ffmpeg.cpp.o
In file included from /usr/include/ni/XnStatus.h:24:0,
                 from /usr/include/ni/XnTypes.h:27,
                 from /usr/include/ni/XnOpenNI.h:27,
                 from /usr/include/ni/XnCppWrapper.h:27,
                 from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20,
                 from /home/chaiein/softwares/Pangolin/src/video/video.cpp:44:
/usr/include/ni/XnPlatform.h:73:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
  #error OpenNI Platform Abstraction Layer - Unsupported Platform!
   ^
In file included from /usr/include/ni/XnTypes.h:28:0,
                 from /usr/include/ni/XnOpenNI.h:27,
                 from /usr/include/ni/XnCppWrapper.h:27,
                 from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20,
                 from /home/chaiein/softwares/Pangolin/src/video/video.cpp:44:
/usr/include/ni/XnOS.h:51:3: error: #error OpenNI OS Abstraction Layer - Unsupported Platform!
  #error OpenNI OS Abstraction Layer - Unsupported Platform!
   ^
In file included from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20:0,
                 from /home/chaiein/softwares/Pangolin/src/video/video.cpp:44:
/usr/include/ni/XnCppWrapper.h:10045:76: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
    XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);
                                                                            ^
/usr/include/ni/XnCppWrapper.h:10104:75: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
   XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);
                                                                           ^
In file included from /usr/include/ni/XnStatus.h:24:0,
                 from /usr/include/ni/XnTypes.h:27,
                 from /usr/include/ni/XnOpenNI.h:27,
                 from /usr/include/ni/XnCppWrapper.h:27,
                 from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20,
                 from /home/chaiein/softwares/Pangolin/src/video/drivers/openni.cpp:28:
/usr/include/ni/XnPlatform.h:73:3: error: #error OpenNI Platform Abstraction Layer - Unsupported Platform!
  #error OpenNI Platform Abstraction Layer - Unsupported Platform!
   ^
In file included from /usr/include/ni/XnTypes.h:28:0,
                 from /usr/include/ni/XnOpenNI.h:27,
                 from /usr/include/ni/XnCppWrapper.h:27,
                 from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20,
                 from /home/chaiein/softwares/Pangolin/src/video/drivers/openni.cpp:28:
/usr/include/ni/XnOS.h:51:3: error: #error OpenNI OS Abstraction Layer - Unsupported Platform!
  #error OpenNI OS Abstraction Layer - Unsupported Platform!
   ^
In file included from /home/chaiein/softwares/Pangolin/include/pangolin/video/drivers/openni.h:20:0,
                 from /home/chaiein/softwares/Pangolin/src/video/drivers/openni.cpp:28:
/usr/include/ni/XnCppWrapper.h:10045:76: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
    XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);
                                                                            ^
/usr/include/ni/XnCppWrapper.h:10104:75: error: macro "XN_VALIDATE_NEW" passed 4 arguments, but takes just 2
   XN_VALIDATE_NEW(pTrans, StateChangedCallbackTranslator, handler, pCookie);

amiltonwong, 2016-03-29:
It seems you still couldn't compile Pangolin successfully.

I had tried the following commands with no failure. 

>> git clone https://github.com/stevenlovegrove/Pangolin.git
>> cd Pangolin
>> mkdir build
>> cd build
>> cmake -DCPP11_NO_BOOST=1 ..
>> make -j

ramtinkr, 2016-06-03:
Hello everyone,

I have somehow the same problem. I'd like to run ORB-SLAM2 on Linux OS (Ubuntu 16.06 Xenial) with CPU intel Corei7 , 2.2 GHZ and I have tried to install the prerequisities, which were written in the README.md file on github.com. But the problem is I have faced several errors during building ORB_SLAM2 and compiling ORB_SLAM2 Examples for mono_tum.cc and mono_kitti.cc. It would be so kind if you can help me in this way because the type of error is somehow the same as above but I have built Pangolin
[image]
[image]
[image]
[image]

 successfully.

ywj447, 2016-08-26:
Hi everyone! I met the same problem as @chaiein . Did you clone the Pangolin into your system for many times and i did, so I reserve only one piece and delete rest of them. Then, compile it like following:
git clone https://github.com/stevenlovegrove/Pangolin.git
cd Pangolin
mkdir build
cd build
cmake -DCPP11_NO_BOOST=1 ..
make -j1
Then, 
cd ORB_SLAM2
chmod +x build.sh
./build.sh
wish it could be helpful

chaiein, 2016-08-26:
I solved this problem after I got virtual and operating system reinstalled. I again used the ORB_SLAM2 freshely it worked with out any problem.

chaiein, 2017-01-10:
Edit the Pangolin openni.cpp files removed the code related to openni. This will also help in successful building of the Pangolin

closed 2017-01-16 by chaiein

riccitensor, 2017-12-11:
Pangolin was compiled already, but unfortunately not well enough. I solved it using https://github.com/raulmur/ORB_SLAM2/issues/22