raulmur/ORB_SLAM2 #24: Can not compile g20

liruihao, 2016-02-18:
I can not compile the g2o version in ORB_SLAM2. It appears blow.

CMake Error at cmake_modules/FindBLAS.cmake:393 (message):
  A required library with BLAS API not found.  Please specify library
  location.
Call Stack (most recent call first):
  CMakeLists.txt:46 (FIND_PACKAGE)

-- Configuring incomplete, errors occurred!

When I use the lastest version download from g2o. I can compile it, but orb_slam2 seems use an different version of g20 so that orb_slam can not find some head file.

What should I do ?

liruihao, 2016-02-18:
I did not install blas and lapack, I have read the readme.txt in ORB_slam for old version. Problem has been solved. You can close it. 

Thank you very much!

LavrentyLavrov, 2016-02-21:
Stumbled upon this too. This should be in readme.

raulmur, 2016-02-22:
BLAS and LAPACK are mentioned in the list of dependencies (Dependencies.md), but I missed to write it on the prerequisites for installation. Thanks!

closed 2016-02-22 by raulmur