raulmur/ORB_SLAM2 #33: Global bundle adjustment causing map to jump

antithing, 2016-03-04:
Hi, thank you for this code, it is brilliant. I have it running live with a stereo camera setup, and at first, the map and translation values look great. But, when it runs GBA, the camera pose suddenly jumps, moving up to a metre from where it was. Is this normal? A stable and accurate camera track is more important to me than map accuracy, could i turn off the Global Bundle adjustment? Or would this be not advised?

Thanks again!

raulmur, 2016-03-04:
Your question is very interesting. Let me explain what is happening.
 
Camera pose can experience a jump, if the MapPoints that are used for tracking the camera, suffer a significant correction. When a loop is closed two optimizations are performed:

1) Essential Graph Optimization: This is a pose graph optimization that distributes the error in the relative pose between the current KeyFrame and a loop closing KeyFrame (an old KeyFrame in your map). In this optimization, the loop closing KeyFrame is fixed, and the rest of KeyFrames are corrected. After the optimization, the MapPoints are corrected according to the corrections of their reference KeyFrames. That means that MapPoints seen by current KeyFrame (and very likely seen by your current camera) will be corrected, as the current KeyFrame pose will be corrected. A way to avoid a jump on the camera, is to fix the current KeyFrame instead of the loop closing KeyFrame. You can change this in line: https://github.com/raulmur/ORB_SLAM2/blob/master/src/Optimizer.cc#L834-L835 , so that the surroundings of the current KeyFrame are not moved by the optimization.

2) Global bundle adjustment corrects all KeyFrames and all MapPoints fixing the initial KeyFrame (the KeyFrame with id 0). Again to avoid a change in the surroundings of your current KeyFrame, fix the current KeyFrame instead of the initial KeyFrame: https://github.com/raulmur/ORB_SLAM2/blob/master/src/Optimizer.cc#L79 . To do this, you will need to add an argument to the Bundle Adjustment function to tell which KeyFrame has to be fixed.

antithing, 2016-03-04:
Thank you! That makes sense. So I should just change:

if(pKF==pLoopKF)
            VSim3->setFixed(true);

to:

VSim3->setFixed(true);

and change:

 vSE3->setFixed(pKF->mnId==0);

to :+1: 
 vSE3->setFixed(pKF->mnId==1);

Is this right?

antithing, 2016-03-07:
Thank you! That makes sense. So for the  Essential Graph Optimization, I need to:

 if(pKF==currentKeyframe)
            VSim3->setFixed(true);

and for the Global:

vSE3->setFixed(currentKeyframe);

Is this correct?

Is there a function existing to find the current camera keyframe? Or is it the last entry in the keyframes list?

Thank you again, very much appreciated.

raulmur, 2016-03-07:
You already know which is the keyframe for which you detect the loop. This is the keyframe with index nLoopKF in BundleAdjustment function. However as the Global BA can take a significant time you will be fixing a keyframe that is far away from your current camera an therefore the camera can still experience a jump. You have two options here:
- The update of the map with the Global BA output is performed here: https://github.com/raulmur/ORB_SLAM2/blob/master/src/LoopClosing.cc#L672-L733 . You could compute the correction for the current reference keyframe (for camera tracking) and then transform all corrections so that the current keyframe remains at the same pose. You will need to implent a function in Tracking that return the current reference keyframe, and acces this by the mpTracker pointer in LoopClosing.

- If you are not super interested in accuracy, you could avoid launching Global BA here: https://github.com/raulmur/ORB_SLAM2/blob/master/src/LoopClosing.cc#L575

antithing, 2016-03-07:
Awesome. Thank you yet again for your help. One more annoying question...

I am trying to get the camera pose in world coordinates from the cameraPose that is returned from TrackStereo. I have borrowed the code from MapDrawer:

void invertCamPose(cv::Mat mCameraPose, pangolin::OpenGlMatrix &M)
{
	if (!mCameraPose.empty())
	{
		cv::Mat Rwc(3, 3, CV_32F);
		cv::Mat twc(3, 1, CV_32F);
		{
			Rwc = mCameraPose.rowRange(0, 3).colRange(0, 3).t();
			twc = -Rwc*mCameraPose.rowRange(0, 3).col(3);
		}

		M.m[0] = Rwc.at<float>(0, 0);
		M.m[1] = Rwc.at<float>(1, 0);
		M.m[2] = Rwc.at<float>(2, 0);
		M.m[3] = 0.0;

		M.m[4] = Rwc.at<float>(0, 1);
		M.m[5] = Rwc.at<float>(1, 1);
		M.m[6] = Rwc.at<float>(2, 1);
		M.m[7] = 0.0;

		M.m[8] = Rwc.at<float>(0, 2);
		M.m[9] = Rwc.at<float>(1, 2);
		M.m[10] = Rwc.at<float>(2, 2);
		M.m[11] = 0.0;

		M.m[12] = twc.at<float>(0);
		M.m[13] = twc.at<float>(1);
		M.m[14] = twc.at<float>(2);
		M.m[15] = 1.0;

		std::cout << twc.at<float>(1) << std::endl;
	}

}

But it does not print the translation value that I would expect (flickers from -00.2 to 00.4). Is there a better approach to get my camera coordinates?

thank you.

antithing, 2016-03-09:
Hi, sorry to keep bothering you... I have made the edits you suggested (without disabling BA) and slam is running great. Thank you! When i measure the tracked distance against a real world ground truth though, the accuracy is quite far off. A movement in Z or X of 1 meter reads as 1.15 meters in the slam system. What could i look at to improve the accuracy of the distance measured? (I am using a Zed stereo camera, with the factory calibrated settings for intrinsics and baseline)

thank you yet again.

raulmur, 2016-03-10:
You need to rectify the stereo images

antithing, 2016-03-10:
Hi,  and thanks again for your time.  The images are synchronised,
rectified and undistorted already.  What else could be my problem?

raulmur, 2016-03-10:
Make sure you are providing correctly the calibration, especially bf parameter (baseline*fx).
At which distance is the camera with respect the scene?

antithing, 2016-03-10:
The camera is around 3 meters from a desk,  with a wall behind it.  I have tried a few different locations around the room and see pretty much the same error.  I am using the camera rigs on board self calibration values,  so will maybe try a new manual opencv stereocalibrate instead...

It seems like even a tiny error in the bf value would throw off the scale,  is this correct?

durn2000, 2016-03-22:
antithing, I am looking at running Orb-SLAM2 with a ZED stereo camera as well and wondering if possibly you could help me this as you may have run through the same issues. Did you perform the opencv stereo calibration? Are you running your camera through ROS and if so which package are you using? Any help you could give me would be appreciated. Perhaps you could email me (my address is in my profile) since this really doesn't belong in this thread.

antithing, 2016-03-22:
Happy to help. stand by for email.

aorait, 2016-06-27:
@antithing @durn2000  I am trying to use ZED camera without ROS. Were you able to do that successfully?

M1234Thomas, 2016-11-29:
hi @antithing @durn2000 , 
I am also working with the ZED stereo camera trying to run live SLAM. But the reconstructions do not look alike the scene. Also the tracking and loop closing are not robust either. I have double checked the settings file and everything seems to be correct. Did you experience something similar in the beginning. Do you mind sharing the rectification code if you have done any and config files as well? Any help is really appreciated. Thanks in advance.

mojovski, 2016-11-29:
Sounds like you are using incorrect camera calibration.

M1234Thomas, 2016-11-29:
In the settings file I can see 2 slightly different set of values for left and the right camera. The doc says camera is factory stereo calibrated. Can someone share the zed configuration file if possible. I am under the impression that the images from the camera are rectified and undistorted. Please correct me if I am wrong about this.

sharma-arjun, 2017-05-01:
@antithing Could you share the changes you had to make to LoopClosing.cc and Tracking.cc to compute the correction for the current reference key frame and transforming all corrections such that the current reference key frame pose does not change (as suggested by raulmur). I am facing the same issue of camera jumping during global bundle adjustment. Thanks