raulmur/ORB_SLAM2 #34: Core dumped running TUM dataset

yizx27, 2016-03-04:
The camera was run for around 5 sec(everything seemed normal), and segmentation fault terminated the program.

I am currently using opencv v2.4.11. I encountered a lot of problems installing opencv library. I manually added #include<iomanip> and #include<opencv2/opencv.hpp> in several files to get the dataset run.

This is the information I got when running TUM dataset:

zixuan@zixuan-MacBookPro:~/ORB_SLAM2$ ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml /home/zixuan/Documents/dataset/rgbd_dataset_freiburg1_xyz

ORB-SLAM2 Copyright (C) 2014-2016 Raul Mur-Artal, University of Zaragoza.
This program comes with ABSOLUTELY NO WARRANTY;
This is free software, and you are welcome to redistribute it
under certain conditions. See LICENSE.txt.

Input sensor was set to: Monocular

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 517.306
- fy: 516.469
- cx: 318.643
- cy: 255.314
- k1: 0.262383
- k2: -0.953104
- k3: 1.16331
- p1: -0.005358
- p2: 0.002628
- fps: 30
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

init done 
opengl support available 
New Map created with 90 points
Segmentation fault (core dumped)

There is no other information provided.
Thank you for help!
P.S. I am using ubuntu14.04 (dual boot) on Macbook pro.

dk683, 2016-03-14:
Does the error happen for other cases?
Could you also try other cases (such as stereo, other dataset, or ROS version)?

yizx27, 2016-03-15:
OK, now I am sure it has something to do with the OpenCV library. I just uninstalled opencv 2.4.11 and installed 2.4.9(including editing the bashrc file) and the code works fine! There is no need to change any segment of the code!

closed 2016-03-15 by yizx27