raulmur/ORB_SLAM2 #38: Unable to build the ros examples!

ankitvora7, 2016-03-11:
Hi,

I'm having an issue with building the ROS examples. The first part of the installation works perfectly.i.e I can successfully build the Thirdparty libraries and the examples. But when I try to build the ROS examples, I get this error:

`[rosbuild] Building package ORB_SLAM2
[rosbuild] Error from directory check: /opt/ros/indigo/share/ros/core/rosbuild/bin/check_same_directories.py  /home/ankit/orb_slam_catkin_ws/ORB_SLAM2/Examples/ROS/ORB_SLAM2
1
Traceback (most recent call last):
  File "/opt/ros/indigo/share/ros/core/rosbuild/bin/check_same_directories.py", line 46, in <module>
    raise Exception
Exception
CMake Error at /opt/ros/indigo/share/ros/core/rosbuild/private.cmake:102 (message):
  [rosbuild] rospack found package "ORB_SLAM2" at "", but the current
  directory is
  "/home/ankit/orb_slam_catkin_ws/ORB_SLAM2/Examples/ROS/ORB_SLAM2".  You
  should double-check your ROS_PACKAGE_PATH to ensure that packages are found
  in the correct precedence order.
Call Stack (most recent call first):
  /opt/ros/indigo/share/ros/core/rosbuild/public.cmake:177 (_rosbuild_check_package_location)
  CMakeLists.txt:4 (rosbuild_init)

-- Configuring incomplete, errors occurred!
`
I guess this could be something to do with opencv. Not sure though!
P.S I have all the dependencies installed successfully.

Thanks

zsy372901, 2016-03-11:
Replace **PATH** by the folder where you cloned ORB_SLAM2:
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:**PATH**/ORB_SLAM2/Examples/ROS

ankitvora7, 2016-03-11:
@zsy372901 
I already did that. This is what I added to my bashrc file:
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/ankit/orb_slam_catkin_ws/ORB_SLAM2/Examples/ROS

ankitvora7, 2016-03-11:
Oh I think I know what the issue is. I forgot to source it. My bad. Let me check and get back :) Thanks!

ankitvora7, 2016-03-11:
It's working! I wasn't sourcing it!

zsy372901, 2016-03-11:
congratulations!

ramtinkr, 2016-06-09:
Hello, @ankitvora7  @zsy372901 @raulmur 

I have got the error of rosbuilt_init while operating cmake .. -DROS_BUILD_TYPE=Release in /Examples/ROS/ORB_SLAM2/build. Here my ROS path in .bashrc file:

echo "source /opt/ros/indigo/setup.bash" >> ~/.bashrc source ~/.bashrc

export Examples/ROS/ORB_SLAM2=${Examples/ROS/ORB_SLAM2}:/home/ramtin/ORB_SLAM2/Examples/ROS

source /opt/ros/indigo/setup.bash

--------------------------------------------------------------------Here is the Error :

ramtin@ramtin:~/ORB_SLAM2/Examples/ROS/ORB_SLAM2$ mkdir build ramtin@ramtin:~/ORB_SLAM2/Examples/ROS/ORB_SLAM2$ cd build ramtin@ramtin:~/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ cmake .. -DROS_BUILD_TYPE=Release -- The C compiler identification is GNU 4.8.5 -- The CXX compiler identification is GNU 4.8.5 -- Check for working C compiler: /usr/bin/cc -- Check for working C compiler: /usr/bin/cc -- works -- Detecting C compiler ABI info -- Detecting C compiler ABI info - done -- Check for working CXX compiler: /usr/bin/c++ -- Check for working CXX compiler: /usr/bin/c++ -- works -- Detecting CXX compiler ABI info -- Detecting CXX compiler ABI info - done CMake Error at CMakeLists.txt:4 (rosbuild_init): Unknown CMake command "rosbuild_init".

-- Configuring incomplete, errors occurred! See also "/home/ramtin/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/CMakeFiles/CMakeOutput.log". ramtin@ramtin:~/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build$ make -j make: * No targets specified and no makefile found. Stop.
[image]
[image]

crvogt, 2016-08-04:
Hi ankitvora7,

I'm having the same problem here; what was it you weren't sourcing?

Thanks!

ankitvora7, 2016-08-04:
Source the catkin workspace where you've kept your orb_slam packages. So in my case, I sourced the setup.bash in the devel folder of "orb_slam_catkin_ws". See if that works!

crvogt, 2016-08-04:
You know I'd swear I sourced it. I guess 3am might not be the best time to try and link everything :D But yeah, that worked, brought up an easy to fix new bug, and now I'm up and running! Thanks for your quick reply.

ankitvora7, 2016-08-04:
Awesome :)

closed 2016-08-04 by ankitvora7

pipi1226, 2016-09-29:
@ramtinkr  
It seems that ros didn't use a workspace. I've met this problem about rosbuilt_init error.
http://wiki.ros.org/catkin/Tutorials/using_a_workspace
May it help you. 
echo $ROS_PACKAGE_PATH
" ROS_PACKAGE_PATH=/opt/ros/indigo/share:/opt/ros/indigo/stacks:PATH_TO_ORB-SLAM2/ORB_SLAM2/Examples/ROS "

It may solve the problem.

Arashm12, 2017-02-04:
@ankitvora7 
I sourced but it doesn't work. What is wrong?
Its my .bashrc:
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/arash/Desktop/ORB_SLAM2/Examples/ROS
source /opt/ros/kinetic/setup.bash

fanyuzeng, 2017-02-17:
@Arashm12   source the setup.bash in ROS file, for example , it's my source:
 source /home/z/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel/setup.bas
good luck!

BW25, 2018-09-12:
Hello, I am getting the same error.
When I attempt to run build_ros.sh, it fails at cmake .. -DROS_BUILD_TYPE=Release

I added this line to .bashrc as instructed
export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/h/software/ORB_SLAM2/Examples/ROS

I noticed the lines about sourcing, and added this line to .bashrc
source /home/h/ORB_SLAM2/Examples/ROS/ORB_SLAM2/build/devel/setup.bash

However, I still get the same error, and when I checked, the devel folder does not exist.

Can anyone help?

BW25, 2018-09-18:
> It's working! I wasn't sourcing it!

If you don't mind, what do you mean by sourcing it, and how can I do this?
Thanks.

BW25, 2018-09-18:
Solved the problem. I followed a guide for installing ROS that may have been a little out of date instead of the official documentation, so I had a few missed dependencies.
If it helps, it was this line:
sudo apt-get install python-rosinstall python-rosinstall-generator python-wstool build-essential
from here:
http://wiki.ros.org/melodic/Installation/Ubuntu

RafaelAgular, 2020-07-28:
> Oh, creo que sé cuál es el problema. Olvidé buscarlo. Culpa mía. Déjame comprobar y volver :) ¡Gracias!
Como lo buscaste?