raulmur/ORB_SLAM2 #39: Pangolin X11: Invalid GLX version. Require GLX >= 1.3

zsy372901, 2016-03-11:
My work circumstance is Windows8.1, so I use VNC View to connect a computer installed ubuntu 14.04 to test ORB_SLAM2 project. However, after I run the command:"rosrun ORB_SLAM2 RGBD ORBvoc.txt settings.yaml", terminal displays these follewing warning:
terminate called after throwing an instance of 'std::runtime_error'
  what():  Pangolin X11: Invalid GLX version. Require GLX >= 1.3
Aborted (core dumped)

ManyuChang, 2016-07-05:
Have you solved the issues?

zsy372901, 2016-08-02:
@ManyuChang 
You can use "nomachine" as the desktop viewer

kongan, 2016-08-23:
@zsy372901 
can you say more clearly ?

fabrice-ducos, 2021-08-14:
You can use the glxinfo command line tool to check which versions of the GLX API are available for your server and client machines. You must ensure that both endpoints support GLX >= 1.3. Especially, if you use the free version of Xming as your X-Server on Windows, you will never get more than GLX 1.2. You can try to replace it with VCXSRV, also free and open-source, but updated for GLX 1.4.

You can also look at this link:

https://github.com/sxyu/volrend/issues/8#issuecomment-898958107