raulmur/ORB_SLAM2 #40: Is Monocular ORB-SLAM in True-Scale?

dk683, 2016-03-14:
Hello. This might be a trivial question, but in the case of monocular ORB-SLAM, is the camera trajectory and 3D reconstruction in true-scale (i.e., if a camera moved 10 meters, the camera trajectory in ORB-SLAM is also around 10 meters)?
I appreciate your time!

MarcGyongyosi, 2016-03-14:
I suggest you read the paper and do some research on monocular SLAM. 

Unless you give some kind of external reference for scale estimation, it is not true scale.

dk683, 2016-03-14:
Thank you for your reply, Marc.
The paper (http://arxiv.org/pdf/1502.00956.pdf) mentioned that it aligned the keyframe trajectories to the ground truth by applying the similarity transformation (Experiments section E).
Also my experiments with my personal dataset with ORB-SLAM shows that monocular ORB-SLAM is not true-scale.
Somehow ... I thought the monocular ORB-SLAM is in true-scale haha
Thank you for the clarification!

closed 2016-03-14 by dk683