raulmur/ORB_SLAM2 #42: My OS (Ububti)

Sfaxi-Rabie, 2016-03-17:


svepe, 2016-03-17:
I have built it on Ubuntu Server 14.04 LTS without any issues. Could you put the output that you get when it crashes?

Sfaxi-Rabie, 2016-03-19:
@svepe Hi Svetlin,
My OS crashes always at this level while building ORB-SLAM2 (check the photo)
[image]

svepe, 2016-03-21:
What do you mean by the OS crashes?

MathewDenny, 2016-03-22:
Will following this discussion fix your problem? #8 

or try manually invoking 
_make -j4_
at your ORB_SLAM2/build location, assuming 4 cores in your device

Sfaxi-Rabie, 2016-03-22:
@svepe I means that every thing stop working, i can't control any thing (i can't even move the mouse curser it blocks) thus i need to shutdown my computer manually by forcing power button.

Sfaxi-Rabie, 2016-03-22:
@MathewDenny My computer is 4 GB RAM and 4 Cores CPU, i don't think that may make a problem, I will try your solution anyway, thinks.

weidi-chang, 2016-04-04:
Have you fixed this issue ?
I got around it by editing the build.sh script and removing the -j option to disable parallel compilation entirely

Ashwinning, 2016-04-09:
I compiled `ORB_SLAM2` on a VM with a 4 core 4GB ram config, it took forever to compile, got stuck there, but eventually resumed and compiled.
On a VM with 6 cores and 8GB ram, compilation was **way** faster, and didn't seem to hang at any point.
Disabling `-j` should help you avoid any hangs/crashes.

ramtinkr, 2016-06-03:
Hello everyone,

I'd like to run ORB-SLAM2 on Linux OS (Ubuntu 16.06 Xenial) with CPU intel Corei7 , 2.2 GHZ and I have tried to install the prerequisities, which were written in the README.md file on github.com. But the problem is I have faced several errors during building ORB_SLAM2 and compiling ORB_SLAM2 Examples for mono_tum.cc and mono_kitti.cc. It would be so kind if you can help me in this way.
[image]
[image]
[image]
[image]

ujjwalrathod007, 2018-06-12:
I also have the exactly same issue..

kouui, 2019-01-31:
Is this issue still alive? 
I have almost the same error when `catkin build` orb_slam2_ros with Ubuntu 18.04.01LTS, ROS/melodic.

> Hello everyone,
>
> I'd like to run ORB-SLAM2 on Linux OS (Ubuntu 16.06 Xenial) with CPU intel Corei7 , 2.2 GHZ and I have tried to install the prerequisities, which were written in the README.md file [...]