raulmur/ORB_SLAM2 #43: unable to run the ros example

kilimeters, 2016-03-18:
Hi
When  running the ros example,i got some problems.
I enter follow command
1.cd ORB_SLAM2&& cd Examples&& cd ROS&&cd ORB_SLAM2
2.export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/cm/ORB_SLAM2/Examples/ROS
3.rosrun ORB_SLAM2 RGBD /home/**/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/**/ORB_SLAM2/Examples/RGB-D/TUM1.yaml

I get this error：
[ERROR] [1458306383.881071390]: [registerPublisher] Failed to contact master at [localhost:11311].  Retrying...
Hope to your reply, thank you.

MathewDenny, 2016-03-22:
Make sure you are running ROS master before running any ROS node. 
rosrun roscore

If the ROS master is running in another machine check your check/set your ROS_IP and ROS_MASTER_URI environment variables.

export ROS_IP=<your IP>;export ROS_MASTER_URI=http://<MASTER IP>:11311;export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/<pathtoORB_SLAM>/Examples/ROS

kilimeters, 2016-03-25:
I just turn on a terminal and enter the last orders:
1.cd ORB_SLAM2&& cd Examples&& cd ROS&&cd ORB_SLAM2
 2.export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/cm/ORB_SLAM2/Examples/ROS
 3.rosrun ORB_SLAM2 
What's the ROS master?
PS:could u leave the email adress?

kilimeters, 2016-03-25:
I just turn on a terminal and enter the last orders:
1.cd ORB_SLAM2&& cd Examples&& cd ROS&&cd ORB_SLAM2
2.export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/cm/ORB_SLAM2/Examples/ROS
3.rosrun ORB_SLAM2
What's the ROS master?
PS:could u leave the email adress?

export ROS_IP=;export ROS_MASTER_URI=http://:11311;export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}://Examples/ROS

MathewDenny, 2016-03-28:
The expected sequence would be
1.roscore &
2. Run the camera node / dataset
2.cd ORB_SLAM2 && cd Examples && cd ROS && cd ORB_SLAM2 
3.export ROS_PACKAGE_PATH=${ROS_PACKAGE_PATH}:/home/cm/ORB_SLAM2/Examples/ROS 4.rosrun ORB_SLAM2 RGBD /home/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/ORB_SLAM2/Examples/RGB-D/TUM1.yaml

The roscore command, which starts the rosmaster,  should always be run first before executing any ROS node. Then only will you be able to communicate between your nodes and also a host of other tasks.

I would request you to get the basic intro to ROS first and then continue on.

zyffse, 2016-04-10:
Hello, MathewDenny, Thanks for your helpful steps. About the second step: run the camera node /dataset, I got some trouble. After1 and 2,3,4, I open a new terminal with: rosbag play --pause rgbd.bag /cam0/image_raw:=/camera/image_raw
There ORB_SLAM2 still is black and waiting for images. Do you know how to fix this. Thank you very much.