raulmur/ORB_SLAM2 #46: ORB-SLAM2: Current Frame gives black screen with "WAITING FOR IMAGES"

sunghoon031, 2016-03-22:
I followed the instruction and was successfully able to run the monocular example with 

./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUMX.yaml PATH_TO_SEQUENCE_FOLDER

Now, I am trying to run monocular example on ROS, so I do the following
- Terminal 1: roscore
- Terminal 2: rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/TUM2.yaml
- Terminal 3: (after cd to directory where .bag file is) rosbag play rgbd_dataset_freiburg2_desk.bag

After this I still see nothing on ORB-SLAM2:Map Viwer and black screen on ORB-SLAM2:Current Frame with "WAITING FOR IMAGES" at the bottom.

I am an absolute ROS beginner, can you please explain how I can get it working?

poine, 2016-03-22:
look on which topic rgbd_dataset_freiburg2_desk.bag has the images
(rostopic list), then remap that topic (
http://wiki.ros.org/Remapping%20Arguments) to /camera/image_raw (that is
where orbslam expects its images)

sunghoon031, 2016-03-22:
Thank you very much !! i was able to run it successfully. For the other noobs like me, this is what I did:

1. rosbag play rgbd_dataset_freiburg2_desk.bag
2. Open another terminal then "rostopic list"
3. The image is published in the topic "/camera/rgb/image_color"
4. Stop the rosbag, then re-run it with "rosbag play rgbd_dataset_freiburg2_desk.bag /camera/rgb/image_color:=/camera/image_raw"

closed 2016-03-22 by sunghoon031