raulmur/ORB_SLAM2 #47: Confusions about opencv version

drinkingcoder, 2016-03-25:
"
Could not find a configuration file for package "OpenCV" that is compatible
  with requested version "2.4.3".

  The following configuration files were considered but not accepted:

    /usr/local/share/OpenCV/OpenCVConfig.cmake, version: 3.1.0
"
When I'm running build.sh, it shows me the message above, but I can't find opencv 2.4.3 package

amiltonwong, 2016-03-29:
What's the output when u issue this command:
$ pkg-config --modversion opencv

Mine is opencv 2.4.8, and I have no problem in compiling orb_slam

chaiein, 2016-04-29:
Even I have the same error. 
$ pkg-config --modversion opencv
3.1.0 is the output

gunshi, 2016-05-06:
Hi, I am trying to compile ORB_SLAM on OSX 10.11.12 with ROS Indigo. i have opencv 3.1 installed (built form source) under /usr/local/lib and /usr/local/include (directories searched by default), and also opencv 2.4.12 under /usr/local/Cellar(installed through brew package manager), which I can link to, in case opencv 3.1 is not compatible.
How can I provide the path to the usr/local/Cellar opencv(2.4.12) instead of the default /usr/local/lib opencv(3.1), in the sense that what variables/flags do i provide in the cmake file/with the cmake command on the terminal?
I tried some experiments with "CPPFLAGS" but it didn't seem to work.
I still got the error:

> CMake Error at CMakeLists.txt:31 (find_package):
>   Could not find a configuration file for package "OpenCV" that is compatible
>   with requested version "2.4.3".
> 
>   The following configuration files were considered but not accepted:
> 
>     /usr/local/share/OpenCV/OpenCVConfig.cmake, version: 3.1.0

 
Thanks!

maxb91, 2016-06-16:
Hi, I've just installed OpenCV from scratch and I've got the same problem when running build.sh. Have you already found a solution to this issue?

yilun, 2016-06-19:
@maxb91 I passed this issue by uninstall opencv 3 , ROS should have installed opencv 2.