raulmur/ORB_SLAM2 #48: weird errors when building DBoW2

houz42, 2016-03-27:
Getting these error messages when building DBoW2 (Using build.sh provided or cmake under /Thirdparty/DBoW2/)
```
Linking CXX shared library ../lib/libDBoW2.so
/usr/bin/ld: cannot find -lQt5::Core
/usr/bin/ld: cannot find -lQt5::Gui
/usr/bin/ld: cannot find -lQt5::Widgets
/usr/bin/ld: cannot find -lQt5::Test
/usr/bin/ld: cannot find -lQt5::Concurrent
/usr/bin/ld: cannot find -lQt5::OpenGL
collect2: error: ld returned 1 exit status
make[2]: *** [../lib/libDBoW2.so] Error 1
make[1]: *** [CMakeFiles/DBoW2.dir/all] Error 2
make: *** [all] Error 2
```
Why lQt5, it dosn't make any sense. I am really confused.

chenyuching, 2016-06-30:
I also have the same problem, but no any idea about that, can anyone help~:（

MCoder1, 2016-08-17:
Any progress? I have the same issue...

houz42, 2016-08-23:
As I remembered, my first builing was failed when I tried to compile the upstream DBoW2 code instead of the modified version in this project. And it successed when I compile the modified code. 

It was too long ago, can't remember too much. hope could be useful.