raulmur/ORB_SLAM2 #53: 'vasprintf' not declared in scope while compiling g2o

Ashwinning, 2016-04-01:
Hi,
I'm compiling ORB_SLAM2 on Windows using `make` (MinGW) and I get the following error:
```
...\ORB_SLAM2\Thirdparty\g2o\g2o\stuff\string_tools.cpp: 
In function 'std::string g2o::formatString(const char*, ...)':
...\ORB_SLAM2\Thirdparty\g2o\g2o\stuff\string_tools.cpp:100:49: 
error: 'vasprintf' was not declared in this scope
   int numChar = vasprintf(&auxPtr, fmt, arg_list);
                                                 ^
...\ORB_SLAM2\Thirdparty\g2o\g2o\stuff\string_tools.cpp: 
In function 'int g2o::strPrintf(std::string&, const char*, ...)':
...\ORB_SLAM2\Thirdparty\g2o\g2o\stuff\string_tools.cpp:117:50: 
error: 'vasprintf' was not declared in this scope
   int numChars = vasprintf(&auxPtr, fmt, arg_list);
                                                  ^
make[2]: *** [CMakeFiles/g2o.dir/g2o/stuff/string_tools.cpp.obj] Error 1
make[1]: *** [CMakeFiles/g2o.dir/all] Error 2
make: *** [all] Error 2
```

I was wondering if anyone else came across this error / if anyone could give me some pointers on how to fix this.

Thanks!

Ashwinning, 2016-04-01:
#HackAlert : Commenting out the `WINDOWS` preprocessor directive allowed it to compile : https://github.com/Ashwinning/ORB_SLAM2/commit/b713744fb221e2a5dc8523cbb8bc1eceb651b20a

I'll take a look at what's causing it to go out of scope and report back when I get the chance.

Maybe it's happening because I'm using MinGW?

closed 2016-04-09 by Ashwinning

makpune, 2021-01-16:
> #HackAlert : Commenting out the `WINDOWS` preprocessor directive allowed it to compile : [Ashwinning@b713744](https://github.com/Ashwinning/ORB_SLAM2/commit/b713744fb221e2a5dc8523cbb8bc1eceb651b20a)
> 
> I'll take a look at what's causing it to go out of scope and report back when I get the chance.
> 
> Maybe it's happening because I'm using MinGW?

Thanks , it worked