raulmur/ORB_SLAM2 #55: Unable to mapping when rotating in a place

JamesChenxie, 2016-04-02:
Hi @raulmur

I was trying to mapping when rotating in a place or in a small circle, but failed.

Is there any limitations to do so in ORB-SLAM2? In Local mapping or in tracking?

I've already commented this lines, but still failed.
//    if(viewCos<viewingCosLimit)
//        return false;

What can I do to achieve my goal?

Hope to your reply, thanks.

JamesChenxie, 2016-04-02:
When rotating beyond a certain angle, the mapping doesn't work well. It can't draw the map point, and then can't tracking.
But in many cases, our robot need to rotate in a place or in a small circle. How can I remove this restriction?
Hope to your reply, thanks.

Ashwinning, 2016-04-11:
MonoSLAM algorithms aren't generally good at tracking pure rotations.
From ORB SLAM 1's [readme](https://github.com/raulmur/ORB_SLAM#7-failure-modes):

> ## Failure Modes
You should expect to achieve good results in sequences similar to those in which we show results in our paper [1], in terms of camera movement and texture in the environment. In general our Monocular SLAM solution is expected to have a bad time in the following situations:

> - No translation at system initialization (or too much rotation).
> - Pure rotations in exploration.
> - Low texture environments.
> - Many (or big) moving objects, especially if they move slowly.

JamesChenxie, 2016-04-15:
I got it. Thank you so much !

closed 2016-04-15 by JamesChenxie