raulmur/ORB_SLAM2 #56: about "Points and keyframe trajetory" and "trajectory and ground truth" plotting

amiltonwong, 2016-04-03:
Dear all,

If I want to re-produce the following plots about "Points and keyframe trajetory" and "trajectory and ground truth", which is illustrated at Fig. 11 in p.13 in ORB_SLAM [paper](http://webdiis.unizar.es/~raulmur/MurMontielTardosTRO15.pdf) , where could I find the corresponding function in ORB_SLAM2 package? Or, if that function does not exist in ORB_SLAM2 package, could someone suggest me how to implement it?

Thanks~
Milton

[image]

nmerrill67, 2017-01-19:
I am currently trying to do the same. If you download the .yaml files from the dataset's website, it will contain transforms from the ground truth tracking frame to the body frame. I believe ORB_SLAM tracks wrt the left camera frame, so you will have to transform to that too. It appears they use Matlab here.

M1234Thomas, 2017-03-07:
Anyone able to plot the trajectories as shown above? I tried to use the TUM tools but the trajectory does not look like what the mapDrawer draws.

nmerrill67, 2017-03-07:
I just used the transformation provided by each dataset's yaml files to transform to the same world coordinates as the ground truth.

gs14iitbbs, 2017-03-17:
Please let me know if anyone exactly how to to the above said comparison.