raulmur/ORB_SLAM2 #57: Why not using OpenCV's ORB extractor?

skylook, 2016-04-05:
Great job for ORB_SLAM2!
But I do have a question that: Why not using OpenCV's ORB extractor?
Cause I see you wrote your own orbextractor, is there anything different than the OpenCV's?
Thanks.

Ashwinning, 2016-04-09:
According to the [readme](https://github.com/raulmur/ORB_SLAM2/blob/master/README.md) : 

> the extraction of ORB is slightly different (trying to improve the dispersion on the image) 

I see that the method signatures are different as well, with the new `minThFAST` variable in `ORB_SLAM2`

`ORB_SLAM`'s `ORBextractor.cc`
```C++
ORBextractor::ORBextractor(int _nfeatures, float _scaleFactor, int _nlevels, 
         int _scoreType, int _fastTh):
    nfeatures(_nfeatures), scaleFactor(_scaleFactor), nlevels(_nlevels),
    scoreType(_scoreType), fastTh(_fastTh)
```
`ORB_SLAM2`'s `ORBextractor.cc`
```C++
ORBextractor::ORBextractor(int _nfeatures, float _scaleFactor, int _nlevels,
         int _iniThFAST, int _minThFAST):
    nfeatures(_nfeatures), scaleFactor(_scaleFactor), nlevels(_nlevels),
    iniThFAST(_iniThFAST), minThFAST(_minThFAST)
```

From the [example yaml file ](https://github.com/raulmur/ORB_SLAM2/blob/master/Examples/Monocular/TUM1.yaml#L38): 
> ORB Extractor: Fast threshold
 Image is divided in a grid. At each cell FAST are extracted imposing a minimum response.
 Firstly we impose iniThFAST. If no corners are detected we impose a lower value minThFAST
 You can lower these values if your images have low contrast

skylook, 2016-04-11:
Thanks for your answer. I can understand it now :)  @Ashwinning

closed 2016-04-14 by skylook