raulmur/ORB_SLAM2 #59: Stuck on `Trying to initialize`.

Ashwinning, 2016-04-07:
Hi!

I have some data I'm trying to run through ORB_SLAM2 but it never goes past the `Trying to Initialize` message.

I tried recording it in different ways - with the video starting with the camera hovering in one place, moving sideways, and moving forward - all with no luck.

I tried increasing `ORBextractor.nFeatures`, `ORBextractor.nLevels:` and decreasing `ORBextractor.iniThFAST` and `ORBextractor.minThFAST` in the YML settings with no luck.

ORB_SLAM2 works fine on the TUM1 (rgbd_dataset_freiburg1_desk) dataset.

Here are the videos I tried to run ORB_SLAM2 on, along with the `.yaml` settings file I'm using : https://drive.google.com/folderview?id=0B8OoRA1AwXH2ZndoRGhWQTRPVDg&usp=sharing

Camera matrix and distortion coefficients were calculated using the OpenCV Calibration example.

I was wondering if someone could give me some pointers on how to fix the issue.

Thanks!
- Ashwin

Beewe, 2016-04-07:
Hello,
I made some quick tests with both ORB_SLAM and ORB_SLAM2 with the videos "out640.mp4" and "output_480.mp4". With increased `ORBextractor.nFeatures` to 2000 I was able to initialize with ORB_SLAM, whereas ORB_SLAM2 didn't work with both videos.

Looking especially at the video "output_480.mp4", I would suggest to turn off the auto-focus of your (smartphone?)-camera and set it to fixed/infinity. With this fixed af you have to do your camera calibration again and also keep it fixed when making new videos. Then test again.

PS: To reduce the motion blur, either try to set your exposure time to a lower value or simply make your test video in a brighter environment. Also keep in mind that ORB_SLAM - no matter what version -  works best with global shutter cameras, whereas your camera surely has rolling shutter. But I also do sometimes tests with rolling shutter cameras and it worked most of the time flawlessly.

Ashwinning, 2016-04-08:
@Beewe 
Thank you so much for the quick response, and thanks for taking the time to run my data through both versions of ORB SLAM, I greatly appreciate it! :smiley: 

I tried recording some more footage using fixed exposure and focus using an iPhone, calibrated the camera, and passed that data into `ORB_SLAM2` - with no luck.

Installing `ORB_SLAM` now to see if I can get stuff working, like you did.
I'll report back once I have it running.

From `ORB_SLAM2`'s [readme](https://github.com/raulmur/ORB_SLAM2/blob/master/README.md) :

> **Notice for ORB-SLAM Monocular users:** ... the extraction of ORB is slightly different (trying to improve the dispersion on the image) and the tracking is also slightly faster. ...

I wonder if that's what's causing this data to not initialize on `ORB_SLAM2`.

Thanks for the pointers on exposure times and global vs rolling shutters!

Ashwinning, 2016-04-09:
Hey @Beewe, any tips on a quick way to feed video files into ORB_SLAM 1?
Thanks!

joyousrabbit, 2016-04-15:
Hi, @Ashwinning,

Could you try this way:

At Initializer::checkRT : 890, put the nGood inside the if{}:
if(cosParallax<0.99998){
vbGood[vMatches12[i].first]=true;
nGood++;
}

Could you tell me if it works? Thanks.

Ashwinning, 2016-04-16:
Thanks for the tip @joyousrabbit !
Haven't gotten around to testing it yet.
I'll report back once I do!

dgsmith, 2016-06-09:
@joyousrabbit @Ashwinning I'm having a very similar problem...I tried @joyousrabbit's fix, but it didn't seem to help :/ Any ideas on this?

joyousrabbit, 2016-06-09:
@dgsmith 
Maybe you can find something on https://github.com/raulmur/ORB_SLAM2/issues/66 . Do you try to initialized by homography (which means by 2d planar such as walls or grounds?)

aorait, 2016-06-17:
Did anyone figure out how to initialize using only homography? @raulmur It doesn't initialize with a planar surface which is required for my application. There is not enough depth in the scene to initialize with the fundamental matrix. The condition "secondBestGood<0.75*bestGood" in Initializer::ReconstructH is never satisfied. Is there any way around it?

odiseo123, 2016-08-17:
I would also be interested in the answers to @aorait question.

hjjayakrishnan, 2017-04-10:
@Ashwinning Hi. Did you get it to initialize in ORB-SLAM2 ?

bakelew4953, 2017-09-12:
@Ashwinning this modification seems to improve initialization, can you suggest why this works?

sunkaianna, 2017-09-12:
@aorait @ odiseo123 @hjjayakrishnan @bakelew4953 I think it's a bug point out by @joyousrabbit. It's very strange that @raulmur Ignored this bug. 

> At Initializer::checkRT : 890, put the nGood inside the if{}:
> if(cosParallax<0.99998){
> vbGood[vMatches12[i].first]=true;
> nGood++;
> }

There is another bug(ORBmatcher.cc, ORBmatcher::SearchForInitialization: 485):

>             float rot = F1.mvKeysUn[i1].angle-F2.mvKeysUn[bestIdx2].angle;
>             if(rot<0.0)
>                 rot+=360.0f;
>             int bin = round(rot*factor);
>             if(bin==HISTO_LENGTH)
>                 bin=0;

should be:

>             float rot = F1.mvKeysUn[i1].angle-F2.mvKeysUn[bestIdx2].angle;
>             if(rot<0.0)
>                 rot+=360.0f;
>             **_int bin = round(rot/(360.0f*factor));_**
>             if(bin==HISTO_LENGTH)
>                 bin=0;

If you modify these bugs, ORBSLAM2 is really wonderful, even on my iPhone  & android devices:)

bakelew4953, 2017-09-14:
From my experience, the modification suggested by @joyousrabbit does generally make initialization happen faster, or more easily. However, the resulting map is not as accurate, and results in more drift, and errors such as double-mapped features. I would suggest not using this feature. In order to initialize correctly, make sure your initial frames have enough movement/parallax to result in a good initialization.

cdlwhm1217096231, 2018-03-20:
@Ashwinning when I am running ORB_SLAM2/ros_mono.cc,but the screen appear "TRYING TO INITIALIZE",i cannot find any reason,please help me,thank you!!!

busybyte1, 2018-06-20:
Windows 8.1, Visual Studio 2017
After endless fails and tries, like calibration, different converts, save and load png's and many more, 
i found out that  

std::chrono::steady_clock::time_point t2 = std::chrono::steady_clock::now();
have to be placed after the main 
myslam.TrackMonocular(RGBMat, tframe); 
call.

I dind't expect this line is so important as i copied parts of example to get a live-video tracking without saving png's, now it works.

If this line doesn't exist, you will hardly get the initialization and if you have success to initialize anyway
i.e.  with ORBextractor.nFeatures: 1500,
the code will freeze while resetting the mapper.
After a stupid fix of the freeze, by uncomment the Line 716 in LocalMapping.cc (//if(!mbResetRequested))  
without the chrono-line , orb slam found and lost initialisation rotatory.
But no tracking was possible.

The calibration parameters are not really important for the initialisation but for a good localisation.
I used stupid parameters and the initialisation and the tracking worked anyway.

i.e.
Camera.fx: 640
Camera.fy: 480
Camera.cx: 320
Camera.cy: 240
Camera.k1: 0
Camera.k2: 0
Camera.p1: 0
Camera.p2: 0
Camera.k3: 0
Camera.fps: 25
Camera.RGB: 1
ORBextractor.nFeatures: 1000
ORBextractor.scaleFactor: 1.8

i hope i could help other frustrated orb slammers cause of this.

Corestars, 2018-07-19:
@sunkaianna thanks bro，after changing as you have mentioned,the time for initialize get short sharply.I thisk that the writer has already changed this bug,but he maybe forget to release,these changing works in Android,in my testing.

fferroni, 2019-05-09:
I tried the code on master, but also the two fixes in this thread, but it still gets stuck on "TRYING TO INITIALIZE" for my data. It's a series of 960x640 images. On the image viewer, there are a lot of keyboards detected though...

yolingen23, 2019-05-14:
> @aorait @ odiseo123 @hjjayakrishnan @bakelew4953 I think it's a bug point out by @joyousrabbit. It's very strange that @raulmur Ignored this bug.
>
>
> There is another bug(ORBmatcher.cc, [...]

my code is ;
`       float rot = F1.mvKeysUn[i1].angle-F2.mvKeysUn[bestIdx2].angle;
                    if(rot<0.0)
                        rot+=360.0f;
                   int bin = round(rot*factor);
                    if(bin==HISTO_LENGTH)
                        bin=0;
                    assert(bin>=0 && bin<HISTO_LENGTH);
                    rotHist[bin].push_back(i1);
                }
            }`  
why is it different from you ?

whyscience, 2019-08-14:
It's useful when distance is large

H-tr, 2022-06-18:
Hi, I got the same problem. I'm trying to run Stereo Node with realsense T265 in the gazebo. 
If I set the camera type as "pinhole", it works (of course, not correct) and gives a result. When I set the camera type to "KannalaBrandt8" and add the corresponding parameters, there keeps showing "Trying to initialize" in the terminal, and no results come out.
I wonder if it is because of an incorrect transformation matrix? But I have verified the T265 intrinsic, and it should be correct. Anyone can help me?

tlee3708, 2025-09-17:
THANK LORD, implementing those two bug changes solved my initialisation issue. I also increased my nFeatures to 4000 and nLevels to 11 for my 1280x960 resolution HIL images.

lx-r, 2025-09-17:
您好，来信已收到，谢谢！