raulmur/ORB_SLAM2 #60: Acquiring pose for All Frames (not Key-Frames)

dk683, 2016-04-08:
Hello.
Through using src/System.cc, I acquired pose (rotation and translation matrix from a world frame) for all key-frames. 
Now I need to acquire the pose for all frames, not key-frames.
I have debugged into src/Frame.cc & src/KeyFrame.cc, but I was not able to find a relevant function or information.
Would this be possible? Or for the system efficiency, does ORB-SLAM only store key-frame poses?
Thank you for your help!

Beewe, 2016-04-08:
Add `cv::Mat pose = SLAM.TrackMonocular(im,tframe); ` in your main for example.

raulmur, 2016-04-08:
Frames are never optimized by bundle adjustment or corrected when closing a loop. The frame pose is only correct at the moment it was computed, you should never use it afterwards. 
You could store the relative pose to a reference keyframe, and compute the frame pose from the keyframe pose. This is done for the stereo setting to recover the full trajectory (see SaveTrajectoryKITTI function is System). However in the monocular setting this would be incorrect, the relative transformation is a rigid body transformation and will not take into account scale corrections.

dk683, 2016-04-23:
Thank you for all the kinds comments :-)

closed 2016-04-23 by dk683

FreakTheMighty, 2016-11-12:
@raulmur the [comment in `SaveTrajectoryTUM`](https://github.com/raulmur/ORB_SLAM2/blob/ab509791f12edead8fbff1f000565af90e8f27ad/src/System.cc#L304-L306) seems to indicate that it is doing what you suggest is already implemented and that all frames would be written out.

>     // Frame pose is stored relative to its reference keyframe (which is optimized by BA and pose graph).
    // We need to get first the keyframe pose and then concatenate the relative transformation.
    // Frames not localized (tracking failure) are not saved.

It seems like its iterating over all relative poses, finding their reference key frame and then writing them to file. Still, I don't see all frames in the output file.

What am I misunderstanding in that code?

sharma-arjun, 2017-05-01:
So, what is the best way to recover the pose for all frames in the monocular setting, if I don't care about the scale? Should I use TrackMonocular (which would have errors as frames are not optimized for bundle adjustment, or should I use relative pose?)

Pratapabhay, 2017-08-12:
@sharma-arjun hye did you find any solution?

arb93, 2017-12-21:
Does anyone have an update on this? Is there a way to save the poses of all frames and not just keyFrames?

It is crucial to my application since the pose data is combined with images to create a volume. Obtaining poses for only keyFrames doesn't accomplish this.

Any thoughts?

yizhou-wang, 2018-07-27:
Is there anyone knows that whether @Beewe 's method works? I can get the poses (monocular) for all frames using ```cv::Mat pose = SLAM.TrackMonocular(im,tframe);``` but I don't really understand why @raulmur mentioned it is incorrect under monocular. Could anyone help me out? Thank you!

hemangchawla, 2019-11-20:
> Frames are never optimized by bundle adjustment or corrected when closing a loop. The frame pose is only correct at the moment it was computed, you should never use it afterwards.
> You could store [...]

Do you mean that the scales of keyframes and non-keyframes would be different as non-keyframes are not optimized? As monocular slam is any way up to scale.