raulmur/ORB_SLAM2 #63: Incomplete behavior? (mono_tum.cc)

varhub, 2016-04-10:
Trying to disable UI of mono_tum.cc it gets stuck (hang) forever.
Exacly same code, config and dataset leads to this dual behavior:
* No UI --> hang at *System Reseting*
* With UI --> same call to *System Reseting* but continues

**problematic datasets**: rgbd_dataset_freiburg1_xyz (x1), rgbd_dataset_freiburg1_360 (x7)
*xN = number of resets with Viewer on (algorithm run normally)*
(more testing in near future)

**bUseViewer = false**
```
ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

New Map created with 82 points
Wrong initialization, reseting...
System Reseting
```

**bUseViewer = true**
```
ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

init done 
opengl support available 

New Map created with 82 points
Wrong initialization, reseting...
System Reseting
Reseting Local Mapper... done
Reseting Loop Closing... done
Reseting Database... done
New Map created with 101 points
-------
```

islers, 2016-04-13:
I can support the observation in that I experienced "random" crashes with the UI disabled. I was running on monocular, live data.

hoangthien94, 2017-11-06:
I have also tried rgbd_dataset_freiburg2_pioneer_slam as well as live monocular data, same problem. Have you found a workaround or solution for this?