raulmur/ORB_SLAM2 #65: different results with rgbd and ros-rgbd

robotsoft, 2016-04-13:
Hi,

I have totally different results with rgbd and ros-rgbd. ros-rgbd looks not working while rgbd looks working well. In ros-rgbd, keyframes are shown at almost same position. I used following commands for rgbd and ros-rgbd respectively.

<rgbd>
$./Examples/RGB-D/rgbd_tum Vocabulary/ORBvoc.txt Examples/RGB-D/TUMX.yaml PATH_TO_SEQUENCE_FOLDER ASSOCIATIONS_FILE

<ros-rgbd>
$rosrun ORB_SLAM2 RGBD Vocabulary/ORBvoc.txt Examples/RGB-D/TUMX.yaml

Thank you.

riematrix, 2016-04-15:
@robotsoft Seems that the ros rgbd example doesn't use the correct topic name. Please check the ros example source code on topic name, at about line 68 or so. Both rgb and depth topic should be modified to match the actual topic name you are using, referring to the result of command "rostopic list".

ameysutavani, 2018-06-13:
@robotsoft were you able to solve the ros_rgbd keyframes being stuck at the same location issue?
I am also facing a similar problem.

robotsoft, 2018-06-14:
@ameysutavani, I'm sorry I didn't have a solution for this issue.

wxyz123456, 2020-07-30:
@robotsoft @ameysutavani  Hi guys, two years have passed, have you solve this problem?

qipengliuNTU, 2020-07-31:
Hi @wxyz123456 . I think I have solved the problem. Please check this page: https://vision.in.tum.de/data/datasets/rgbd-dataset/file_formats#intrinsic_camera_calibration_of_the_kinect. It says that "factor = 5000 # for the 16-bit PNG files
# OR: factor = 1 # for the 32-bit float images in the ROS bag files". So we should change the value of DepthMapFactor in Examples/RGB-D/TUMX.yaml if we want to run ros-rgbd.