raulmur/ORB_SLAM2 #66: [bug] Initializer::ReconstructH doesn't work

joyousrabbit, 2016-04-13:
Hello, I'm not very good at maths. Just after studied <<Deeper understanding of the homography decomposition for vision-based control>> (Ezio Malis, Manuel Vargas,2007), and also do some tests, I think that Initializer::ReconstructH cannot work as expected.

As in the code, one of the conditions to return true from ReconstructH is: secondBestGood<0.75*bestGood

However, in the 8 solutions of Homography decomposition, 2 of thems are completely different but both give a good physic and numerical meaning. Without prior knowledge or the 3rd image, we cannot tell which is the true one. 

The consequence is clear: secondBestGood always equals bestGood. And ReconstructH will always return false.

Simply add inside the cycle " cout<<nGood<<endl; ", by dataset freiburg1_xyz, we can see:

rgb/1305031102.275326.png
rgb/1305031102.311267.png
H
157
3
157
82
0
0
0
0

rgb/1305031102.343233.png
H
63
153
0
153
0
0
0
0

rgb/1305031102.375329.png
H
138
0
138
45
0
0
0
0

rgb/1305031102.411258.png
H
133
0
133
20
0
0
0
0

rgb/1305031102.443271.png
H
107
0
107
5
0
0
0
0

rgb/1305031102.475318.png
H
114
0
[... 1609 lines cut ...]
1
1
0

rgb/1305031109.075410.png
H
75
75
36
0
0
3
3
0

rgb/1305031109.111282.png
H
2
6
75
75
0
0
0
0

raulmur, 2016-04-14:
In our T-RO paper: http://webdiis.unizar.es/~raulmur/MurMontielTardosTRO15.pdf
and in:  " O.  D.  Faugeras  and  F.  Lustman,  “Motion  and  structure  from  motion in  a  piecewise  lanar  environment,” International  Journal  of  Pattern Recognition and Artificial Intelligence
, vol. 2, no. 03, pp. 485–508, 1988.", you will find the answer.

closed 2016-04-14 by raulmur

joyousrabbit, 2016-04-14:
Hello Raulmur. Thank you for response. I read your excellent paper and the related ones. In your paper it mentions "waiting until significant parallax to eliminate the twofold ambiguity". However, according to my test, even the parallax is quite significant, the twofold ambiguity is still there. What's more, it's very long, must wait hundreds of frames to get such significant parallax. 

Why not use 3 frames to do it? Even remove the condition "secondBestGood<0.75*bestGood" and let the following LOST check can initialize the system in several frames instead of hundreds.

joyousrabbit, 2016-04-14:
Or maybe at Initializer::checkRT : 890, it should be:
        if(cosParallax<0.99998){
            vbGood[vMatches12[i].first]=true;
            nGood++;
        }

rikpires, 2016-04-15:
Hi @joyousrabbit ,
I'm not quite sure about the twofold ambiguity, so I cannot answer your question directly. But in my understanding, in ORB-SLAM, the author prefers to retrieve the pose from fundamental matrix which should be success in a significant parallax and sufficient non-coplanar feature matches. In my practice, the initialization is pretty fast in a complex scene.

joyousrabbit, 2016-04-15:
Hi, @rikpires 
That's because your scene is complex and always trigger the Fundmental step. Could you test a planar scene for example against the wall or ground? If I'm right it will never be initialized.

joyousrabbit, 2016-04-15:
Hi, @raulmur 
Why a significant parallax check can tell the true one from the false? The parallax depends the degree of the camera to that planar. If the Z-axis is nearly parallel to the test plane, the little parallax should win (because most points are quite far). Instead, if the z-axis is perpendicular to the test plane, the bigger parallax wins.

odiseo123, 2016-08-17:
Hi @joyousrabbit,
I am working on sequences where a large parallax is difficult to achieve (fulcrum effect). Do you have an update on the changes you made to the initialization step?

Thanks a lot,