raulmur/ORB_SLAM2 #67: Easy to loose tracking on camera

skylook, 2016-04-15:
Thx for the awesome project! But I find when moving faster, it loose tracking easily (monocular). 
Is it a problem of speed? Any ideas on how to improve this? Thx :)

randoms, 2016-04-19:
I think this is hard to improve by code. When you are moving faster, the image will get a very strong motion blur. Also current frame may not share any key points with other key frames.
So a camera with higher fps and a faster computer will improve the performance.

skylook, 2016-04-25:
Thanks for your answer @randoms . I agree with that. 
However, as ORB SLAM relies on FAST corner detection, it's not working well on blur images. Maybe patch tracking on NCC may work a little better when loosing focus?