raulmur/ORB_SLAM2 #68: when using ./build.sh to ORB_SLAM2 ,there occurs  this problem

sakurazhu, 2016-04-15:
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 4
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2

i have tried to restall the ORB_SLAM2, but the errors still exist . i really do not know what to do next ,pleease help me out  ,thank u very much

sakurazhu, 2016-04-15:
@raulmur

joyousrabbit, 2016-04-15:
@sakurazhu 
Could you pose all the compile output? There should be some lines before.

poine, 2016-04-15:
Also  remocing -j from build.sh to make the build sequential rather than
parallel could help debugging

sakurazhu, 2016-04-16:
The before lines are as follow:
PS : I have tried  make -j4 when debugging  this time ,but the error still there.@poine  @joyousrabbit 
`[ 16%] Building CXX object modules/nonfree/CMakeFiles/opencv_perf_nonfree_pch_dephelp.dir/opencv_perf_nonfree_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_perf_ocl_pch_dephelp.a
[ 16%] Built target opencv_perf_ocl_pch_dephelp
[ 16%] Generating opencv_test_nonfree_pch_dephelp.cxx
Scanning dependencies of target opencv_test_nonfree_pch_dephelp
[ 16%] Building CXX object modules/nonfree/CMakeFiles/opencv_test_nonfree_pch_dephelp.dir/opencv_test_nonfree_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_nonfree_pch_dephelp.a
[ 16%] Built target opencv_nonfree_pch_dephelp
[ 16%] Generating opencv_contrib_pch_dephelp.cxx
Scanning dependencies of target opencv_contrib_pch_dephelp
[ 16%] Building CXX object modules/contrib/CMakeFiles/opencv_contrib_pch_dephelp.dir/opencv_contrib_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_test_ocl_pch_dephelp.a
[ 16%] Built target opencv_test_ocl_pch_dephelp
[ 16%] Generating opencv_test_contrib_pch_dephelp.cxx
Scanning dependencies of target opencv_test_contrib_pch_dephelp
[ 16%] Building CXX object modules/contrib/CMakeFiles/opencv_test_contrib_pch_dephelp.dir/opencv_test_contrib_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_perf_nonfree_pch_dephelp.a
[ 16%] Built target opencv_perf_nonfree_pch_dephelp
[ 16%] Generating opencv_stitching_pch_dephelp.cxx
Scanning dependencies of target opencv_stitching_pch_dephelp
[ 17%] Building CXX object modules/stitching/CMakeFiles/opencv_stitching_pch_dephelp.dir/opencv_stitching_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_contrib_pch_dephelp.a
[ 17%] Built target opencv_contrib_pch_dephelp
[ 17%] Generating opencv_perf_stitching_pch_dephelp.cxx
Linking CXX static library ../../lib/libopencv_test_nonfree_pch_dephelp.a
Scanning dependencies of target opencv_perf_stitching_pch_dephelp
[ 17%] Built target opencv_test_nonfree_pch_dephelp
[ 17%] Building CXX object modules/stitching/CMakeFiles/opencv_perf_stitching_pch_dephelp.dir/opencv_perf_stitching_pch_dephelp.cxx.o
[ 17%] Generating opencv_test_stitching_pch_dephelp.cxx
Scanning dependencies of target opencv_test_stitching_pch_dephelp
[ 18%] Building CXX object modules/stitching/CMakeFiles/opencv_test_stitching_pch_dephelp.dir/opencv_test_stitching_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_test_contrib_pch_dephelp.a
[ 18%] Built target opencv_test_contrib_pch_dephelp
[ 18%] Generating opencv_perf_superres_pch_dephelp.cxx
Scanning dependencies of target opencv_perf_superres_pch_dephelp
[ 18%] Building CXX object modules/superres/CMakeFiles/opencv_perf_superres_pch_dephelp.dir/opencv_perf_superres_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_stitching_pch_dephelp.a
[ 18%] Built target opencv_stitching_pch_dephelp
[ 18%] Generating opencv_superres_pch_dephelp.cxx
Scanning dependencies of target opencv_superres_pch_dephelp
[ 19%] Building CXX object modules/superres/CMakeFiles/opencv_superres_pch_dephelp.dir/opencv_superres_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_perf_stitching_pch_dephelp.a
[ 19%] Built target opencv_perf_stitching_pch_dephelp
[ 19%] Generating opencv_test_superres_pch_dephelp.cxx
Scanning dependencies of target opencv_test_superres_pch_dephelp
[ 19%] Building CXX object modules/superres/CMakeFiles/opencv_test_superres_pch_dephelp.dir/opencv_test_superres_pch_dephelp.cxx.o
Linking CXX static library ../../lib/libopencv_test_stitching_pch_dephelp.a
[ 19%] Built target opencv_test_stitching_pch_dephelp
[ 19%] Linking CXX static library ../../lib/libopencv_superres_pch_dephelp.a
Generating opencv_videostab_pch_dephelp.cxx
[ 19%] Built target opencv_superres_pch_dephelp
Scanning dependencies of target libtiff
Linking CXX static library ../../lib/libopencv_perf_superres_pch_dephelp.a
Scanning dependencies of target opencv_videostab_pch_dephelp
[ 19%] Built target opencv_perf_superres_pch_dephelp
[ 20%] [ 20%] Building CXX object modules/videostab/CMakeFiles/opencv_videostab_pch_dephelp.dir/opencv_videostab_pch_dephelp.cxx.o
Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_aux.c.o
Scanning dependencies of target libpng
[ 21%] Building C object 3rdparty/libpng/CMakeFiles/libpng.dir/pngset.c.o
[ 21%] Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_close.c.o
[ 21%] Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_codec.c.o
[ 21%] Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_color.c.o
[ 21%] Building C object 3rdparty/libpng/CMakeFiles/libpng.dir/pngrtran.c.o
[ 21%] Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_compress.c.o
[ 21%] Linking CXX static library ../../lib/libopencv_videostab_pch_dephelp.a
Building C object 3rdparty/libtiff/CMakeFiles/libtiff.dir/tif_dir.c.o
[ 21%] Built target opencv_videostab_pch_dephelp
Scanning dependencies of target IlmImf
[... 325 lines cut ...]
Scanning dependencies of target opencv_core
[ 39%] Built target pch_Generate_opencv_superres
[ 39%] [ 39%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/parallel.cpp.o
Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/gpumat.cpp.o
[ 39%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/matrix.cpp.o
[ 39%] [ 39%] Built target pch_Generate_opencv_videostab
[... cut ...]
/home/sakura/opencv-2.4.10/modules/core/src/system.cpp: In function ‘(static initializers for /home/sakura/opencv-2.4.10/modules/core/src/system.cpp)’:
/home/sakura/opencv-2.4.10/modules/core/src/system.cpp:280:10: error: inconsistent operand constraints in an ‘asm’
make[2]: *** [modules/core/CMakeFiles/opencv_core.dir/src/system.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 39%] Built target pch_Generate_opencv_test_superres
make[1]: *** [modules/core/CMakeFiles/opencv_core.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/opencv-2.4.10/build$ 

- 

- [ ] 

- 

`

joyousrabbit, 2016-04-16:
@sakurazhu 
It's a bug of opencv2.10, you can follow this link to modifiy and rebuild opencv. http://code.opencv.org/issues/3983

If you don't want to build opencv from source code, just type:
sudo apt-get install libcv-dev libcvaux-dev libhighgui-dev libopencv-dev

It works for me always.

sakurazhu, 2016-04-16:
i have tried your method,but  the error happened again  and one more error happened 
@joyousrabbit 

[ 18%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/gpumat.cpp.o
[ 19%] Built target pch_Generate_opencv_videostab
[ 19%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/parallel.cpp.o
[ 19%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/matrix.cpp.o
[ 19%] [ 19%] Built target pch_Generate_opencv_superres
Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/opengl_interop.cpp.o
[ 20%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/system.cpp.o
[ 20%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/persistence.cpp.o
[ 20%] Built target pch_Generate_opencv_test_superres
[ 20%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/matmul.cpp.o
/home/sakura/opencv-2.4.10/modules/core/src/system.cpp: In function ‘(static initializers for /home/sakura/opencv-2.4.10/modules/core/src/system.cpp)’:
/home/sakura/opencv-2.4.10/modules/core/src/system.cpp:280:10: error: inconsistent operand constraints in an ‘asm’
make[2]: *** [modules/core/CMakeFiles/opencv_core.dir/src/system.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [modules/core/CMakeFiles/opencv_core.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/opencv-2.4.10/build$

sakurazhu, 2016-04-16:
i have tried gcc  4.6  and gcc 4.9  ,however ,the error still  there @joyousrabbit  @poine  @raulmur

joyousrabbit, 2016-04-16:
@sakurazhu 
Did you modifified the source code as explained by the forum?

I suggest you try the following steps:
1. supprime the downloaded opencv2.10;
2. sudo apt-get install libcv-dev libcvaux-dev libhighgui-dev libopencv-dev
3. ./build.sh the ORB-slam

sakurazhu, 2016-04-18:
thank u @joyousrabbit . I have managed to solve the problem with opencv by your meathod . However when running ORB-SLAM2  ,new problems come again.

/usr/include/eigen3/Eigen/src/Core/NoAlias.h:95:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator-=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 3, 0, 6, 3>&; Rhs = const Eigen::Matrix<double, 3, 1>&; int NestingFlags = 6; ExpressionType = Eigen::Matrix<double, 6, 1>; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:62:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 3; E = Eigen::Matrix<double, 3, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/System.cc:426:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 1>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256> > >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256> > >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256> > >’
/usr/include/eigen3/Eigen/src/Core/SelfCwiseBinaryOp.h:58:63:   required from ‘class Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256> >’
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:221:84:   required from ‘Derived& Eigen::MatrixBase<Derived>::operator-=(const Eigen::MatrixBase<OtherDerived>&) [with OtherDerived = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256>; Derived = Eigen::Matrix<double, 6, 1>]’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:95:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator-=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 3, 0, 6, 3>&; Rhs = const Eigen::Matrix<double, 3, 1>&; int NestingFlags = 6; ExpressionType = Eigen::Matrix<double, 6, 1>; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:62:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 3; E = Eigen::Matrix<double, 3, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/LoopClosing.cc:770:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 1>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, const Eigen::Matrix<double, 3, 1>&, 256> > >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, Eigen::Map<Eigen::Matrix<double, 3, 6, 0, 3, 6>, 1, Eigen::Stride<0, 0> >, 256> > >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, Eigen::Map<Eigen::Matrix<double, 3, 6, 0, 3, 6>, 1, Eigen::Stride<0, 0> >, 256> > >’
/usr/include/eigen3/Eigen/src/Core/SelfCwiseBinaryOp.h:58:63:   required from ‘class Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, Eigen::Map<Eigen::Matrix<double, 3, 6, 0, 3, 6>, 1, Eigen::Stride<0, 0> >, 256> >’
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:235:77:   required from ‘Derived& Eigen::MatrixBase<Derived>::operator+=(const Eigen::MatrixBase<OtherDerived>&) [with OtherDerived = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, Eigen::Map<Eigen::Matrix<double, 3, 6, 0, 3, 6>, 1, Eigen::Stride<0, 0> >, 256>; Derived = Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >]’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:91:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator+=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const E [... cut ...] 0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >::<anonymous enum>’ [-Wenum-compare]
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/ORB_SLAM2$

sakurazhu, 2016-04-18:
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/ORB_SLAM2$

joyousrabbit, 2016-04-18:
I'm glad that it helps. For the new error, you should be very close because you already got the .o files. So I guess that it's a link problem.

Try do it again, take care of all the parameters behind cmake and make.

sakurazhu, 2016-04-19:
i have tried to do it again  ,but errors were still there. i wonder if there is something wrong with eigen,bacause the report showed that the matrix is not matched. However i  don't konw how to change it 

 @joyousrabbit  @raulmur 

de<0, 0> >; int NestingFlags = 6; ExpressionType = Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:63:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 3; E = Eigen::Matrix<double, 3, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/LoopClosing.cc:770:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 6>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 3, 0, 6, 3>&, Eigen::Map<Eigen::Matrix<double, 3, 6, 0, 3, 6>, 1, Eigen::Stride<0, 0> >, 256> >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 2, 2> > >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 2, 2> > >’
/usr/include/eigen3/Eigen/src/Core/CwiseUnaryOp.h:105:7:   required from ‘class Eigen::CwiseUnaryOpImpl<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 2, 2>, Eigen::Dense>’
/usr/include/eigen3/Eigen/src/Core/CwiseUnaryOp.h:72:7:   required from ‘class Eigen::CwiseUnaryOp<Eigen::internal::scalar_multiple_op<double>, const Eigen::Matrix<double, 2, 2> >’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_edge.h:98:41:   required from ‘g2o::BaseEdge<D, E>::InformationType g2o::BaseEdge<D, E>::robustInformation(const Vector3d&) [with int D = 2; E = Eigen::Matrix<double, 2, 1>; g2o::BaseEdge<D, E>::InformationType = Eigen::Matrix<double, 2, 2>; Eigen::Vector3d = Eigen::Matrix<double, 3, 1>]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:60:66:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 2; E = Eigen::Matrix<double, 2, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/LoopClosing.cc:770:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<2, 2>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::Matrix<double, 2, 2> >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> >’
/usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:126:7:   required from ‘class Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256>’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:95:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator-=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 2, 0, 6, 2>&; Rhs = const Eigen::Matrix<double, 2, 1>&; int NestingFlags = 6; ExpressionType = Eigen::Matrix<double, 6, 1>; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:62:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 2; E = Eigen::Matrix<double, 2, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/LoopClosing.cc:770:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 1>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> >’
/usr/include/eigen3/Eigen/src/Core/products/CoeffBasedProduct.h:126:7:   required from ‘class Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256>’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:91:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator+=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 2, 0, 6, 2>&; Rhs = [... cut ...]  6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >::<anonymous enum>’ [-Wenum-compare]
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/ORB_SLAM2$

joyousrabbit, 2016-04-19:
Which commands did you use? In which path?
Could you show me the history like sakura@ubuntu:~/ORB_SLAM2$XXX XXXXXX  which you typed?

sakurazhu, 2016-04-19:
@joyousrabbit    i used the following commands:

sakura@ubuntu:~ORB_SLAM2  $ chmod +x build.sh
sakura@ubuntu:~ORB_SLAM2  $  ./build.sh

joyousrabbit, 2016-04-19:
@sakurazhu 
So was the executalble in folder example/mono_tum/ generated?
If yes, it's ok, just run it.
if no, I guess you have reinstall eigen3.

sakurazhu, 2016-04-20:
@joyousrabbit  i tried to reinstall eigen3.0 but the error last .   And i did not find  the  executalble in  folder example/mono_tum/  files.   

/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:119:7:   required from ‘class Eigen::CwiseBinaryOp<Eigen::internal::scalar_conj_product_op<double, double>, const Eigen::Matrix<double, 2, 1>, const Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 2, 2>&, const Eigen::Matrix<double, 2, 1>&, 6> >’
/usr/include/eigen3/Eigen/src/Core/Dot.h:47:121:   required from ‘static Eigen::internal::dot_nocheck<T, U, NeedToTranspose>::ResScalar Eigen::internal::dot_nocheck<T, U, NeedToTranspose>::run(const Eigen::MatrixBase<Derived>&, const Eigen::MatrixBase<U>&) [with T = Eigen::Matrix<double, 2, 1>; U = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 2, 2>&, const Eigen::Matrix<double, 2, 1>&, 6>; bool NeedToTranspose = false; Eigen::internal::dot_nocheck<T, U, NeedToTranspose>::ResScalar = double]’
/usr/include/eigen3/Eigen/src/Core/Dot.h:86:71:   required from ‘typename Eigen::internal::scalar_product_traits<typename Eigen::internal::traits<T>::Scalar, typename Eigen::internal::traits<OtherDerived>::Scalar>::ReturnType Eigen::MatrixBase<Derived>::dot(const Eigen::MatrixBase<OtherDerived>&) const [with OtherDerived = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 2, 2>&, const Eigen::Matrix<double, 2, 1>&, 6>; Derived = Eigen::Matrix<double, 2, 1>; typename Eigen::internal::scalar_product_traits<typename Eigen::internal::traits<T>::Scalar, typename Eigen::internal::traits<OtherDerived>::Scalar>::ReturnType = double]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_edge.h:60:47:   required from ‘double g2o::BaseEdge<D, E>::chi2() const [with int D = 2; E = Eigen::Matrix<double, 2, 1>]’
/home/sakura/ORB_SLAM2/src/LocalMapping.cc:762:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<2, 1>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CwiseBinaryOp<Eigen::internal::scalar_conj_product_op<double, double>, const Eigen::Matrix<double, 2, 1>, const Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 2, 2>&, const Eigen::Matrix<double, 2, 1>&, 6> > >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> > >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> > >’
/usr/include/eigen3/Eigen/src/Core/SelfCwiseBinaryOp.h:58:63:   required from ‘class Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> >’
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:221:84:   required from ‘Derived& Eigen::MatrixBase<Derived>::operator-=(const Eigen::MatrixBase<OtherDerived>&) [with OtherDerived = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256>; Derived = Eigen::Matrix<double, 6, 1>]’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:95:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator-=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 2, 0, 6, 2>&; Rhs = const Eigen::Matrix<double, 2, 1>&; int NestingFlags = 6; ExpressionType = Eigen::Matrix<double, 6, 1>; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:62:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 2; E = Eigen::Matrix<double, 2, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/LocalMapping.cc:762:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 1>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CwiseBinaryOp<Eigen::internal::scalar_difference_op<double>, Eigen::Matrix<double, 6, 1>, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, const Eigen::Matrix<double, 2, 1>&, 256> > >::<anonymous enum>’ [-Wenum-compare]
/usr/include/eigen3/Eigen/src/Core/DenseBase.h: In instantiation of ‘class Eigen::DenseBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >’:
/usr/include/eigen3/Eigen/src/Core/MatrixBase.h:61:34:   required from ‘class Eigen::MatrixBase<Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >’
[... cut ...]  6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >::<anonymous enum>’ [-Wenum-compare]
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Tracking.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LocalMapping.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2
sakura@ubuntu:~/ORB_SLAM2$

joyousrabbit, 2016-04-20:
@sakurazhu 
I think it's a g++ bug. 
Could you show me your g++ --version, gcc --version, and your system version?

sakurazhu, 2016-04-21:
@joyousrabbit  yes  the   gcc version  is 4.9.2 (Ubuntu 4.9.2-0ubuntu1~12.04) .
Memory 3.9G
Process  intel core i5 -2430  CPU 2.5hz*4
OS type  32bit 
Disk 20G

sakura@ubuntu:~$ g++ -v
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/lib/gcc/i686-linux-gnu/4.9/lto-wrapper
Target: i686-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 4.9.2-0ubuntu1~12.04' --with-bugurl=file:///usr/share/doc/gcc-4.9/README.Bugs --enable-languages=c,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-4.9 --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --with-gxx-include-dir=/usr/include/c++/4.9 --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-4.9-i386/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-4.9-i386 --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-4.9-i386 --with-arch-directory=i386 --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --enable-objc-gc --enable-targets=all --enable-multiarch --disable-werror --with-arch-32=i686 --with-multilib-list=m32,m64 --enable-multilib --with-tune=generic --enable-checking=release --build=i686-linux-gnu --host=i686-linux-gnu --target=i686-linux-gnu
Thread model: posix
gcc version 4.9.2 (Ubuntu 4.9.2-0ubuntu1~12.04)

joyousrabbit, 2016-04-21:
@sakurazhu 
OK, I konw. 
Please use the command ' ./build.sh > build.txt ' and send the generated build.txt to my email fujunde@gmail.com . We will see what happened.

joyousrabbit, 2016-04-22:
@sakurazhu 
According to your build information, your Eigen3 version is 3.0.5, at least version 3.1.0 is required.

sakurazhu, 2016-04-22:
i have tried to use eigen3.2.8, but it did not work,the same problems appeared again.  @joyousrabbit
Anyway,thank you very much for your unselfish help .
The problems were so stubborn! 

fCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >’
/usr/include/eigen3/Eigen/src/Core/SelfCwiseBinaryOp.h:58:63:   required from ‘class Eigen::SelfCwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> >’
/usr/include/eigen3/Eigen/src/Core/CwiseBinaryOp.h:235:77:   required from ‘Derived& Eigen::MatrixBase<Derived>::operator+=(const Eigen::MatrixBase<OtherDerived>&) [with OtherDerived = Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256>; Derived = Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >]’
/usr/include/eigen3/Eigen/src/Core/NoAlias.h:91:37:   required from ‘ExpressionType& Eigen::NoAlias<ExpressionType, StorageBase>::operator+=(const Eigen::CoeffBasedProduct<Lhs, Rhs, NestingFlags>&) [with Lhs = const Eigen::Matrix<double, 6, 2, 0, 6, 2>&; Rhs = Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >; int NestingFlags = 6; ExpressionType = Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >; StorageBase = Eigen::MatrixBase]’
/home/sakura/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_unary_edge.hpp:63:27:   required from ‘void g2o::BaseUnaryEdge<D, E, VertexXi>::constructQuadraticForm() [with int D = 2; E = Eigen::Matrix<double, 2, 1>; VertexXi = g2o::VertexSE3Expmap]’
/home/sakura/ORB_SLAM2/src/System.cc:426:1:   required from here
/usr/include/eigen3/Eigen/src/Core/DenseBase.h:172:59: warning: enumeral mismatch in conditional expression: ‘Eigen::internal::size_at_compile_time<6, 6>::<anonymous enum>’ vs ‘Eigen::internal::traits<Eigen::CwiseBinaryOp<Eigen::internal::scalar_sum_op<double>, Eigen::Map<Eigen::Matrix<double, 6, 6>, 1, Eigen::Stride<0, 0> >, Eigen::CoeffBasedProduct<const Eigen::Matrix<double, 6, 2, 0, 6, 2>&, Eigen::Map<Eigen::Matrix<double, 2, 6, 0, 2, 6>, 1, Eigen::Stride<0, 0> >, 256> > >::<anonymous enum>’ [-Wenum-compare]
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/LoopClosing.cc.o] Error 1
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/System.cc.o] Error 1
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
make: *** [all] Error 2

joyousrabbit, 2016-04-22:
@sakurazhu 
You're welcome.
Could you show me the new build.txt as generated?

sakurazhu, 2016-04-22:
ok .

2016-04-22 

myworldpig

sakurazhu, 2016-04-22:
ok 

2016-04-22 

myworldpig

Matrixio, 2016-05-14:
I have the same problem.

[ 42%] Building CXX object CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o
In file included from /usr/include/eigen3/Eigen/Core:297:0,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/a/ORB_SLAM2/Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/a/ORB_SLAM2/include/LoopClosing.h:34,
                 from /home/a/ORB_SLAM2/include/Optimizer.h:27,
                 from /home/a/ORB_SLAM2/src/Optimizer.cc:21:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h: In instantiation of ‘void Eigen::internal::call_assignment_no_alias(Dst&, const Src&, const Func&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>]’:
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:712:27:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&, const Func&, typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Func = Eigen::internal::assign_op<int>; typename Eigen::internal::enable_if<(! Eigen::internal::evaluator_assume_aliasing<Src>::value), void*>::type = void*]’
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:693:18:   required from ‘void Eigen::internal::call_assignment(Dst&, const Src&) [with Dst = Eigen::Matrix<int, -1, 1>; Src = Eigen::Matrix<long int, -1, 1, 0, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/PlainObjectBase.h:682:32:   required from ‘Derived& Eigen::PlainObjectBase<Derived>::_set(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; Derived = Eigen::Matrix<int, -1, 1>]’
/usr/include/eigen3/Eigen/src/Core/Matrix.h:225:24:   required from ‘Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>& Eigen::Matrix<_Scalar, _Rows, _Cols, _Options, _MaxRows, _MaxCols>::operator=(const Eigen::DenseBase<OtherDerived>&) [with OtherDerived = Eigen::Matrix<long int, -1, 1, 0, -1, 1>; _Scalar = int; int _Rows = -1; int _Cols = 1; int _Options = 0; int _MaxRows = -1; int _MaxCols = 1]’
/usr/include/eigen3/Eigen/src/Core/PermutationMatrix.h:367:17:   required from ‘Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>& Eigen::PermutationMatrix<SizeAtCompileTime, MaxSizeAtCompileTime, IndexType>::operator=(const Eigen::PermutationBase<OtherDerived>&) [with Other = Eigen::PermutationMatrix<-1, -1, long int>; int SizeAtCompileTime = -1; int MaxSizeAtCompileTime = -1; _StorageIndex = int]’
/home/a/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:68:18:   required from ‘void g2o::LinearSolverEigen<MatrixType>::CholeskyDecomposition::analyzePatternWithPermutation(g2o::LinearSolverEigen<MatrixType>::SparseMatrix&, const PermutationMatrix&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>; g2o::LinearSolverEigen<MatrixType>::SparseMatrix = Eigen::SparseMatrix<double, 0>; g2o::LinearSolverEigen<MatrixType>::PermutationMatrix = Eigen::PermutationMatrix<-1, -1, long int>]’
/home/a/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:195:9:   required from ‘void g2o::LinearSolverEigen<MatrixType>::computeSymbolicDecomposition(const g2o::SparseBlockMatrix<MatrixType>&) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/a/ORB_SLAM2/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h:100:37:   required from ‘bool g2o::LinearSolverEigen<MatrixType>::solve(const g2o::SparseBlockMatrix<MatrixType>&, double*, double*) [with MatrixType = Eigen::Matrix<double, 7, 7, 0, 7, 7>]’
/home/a/ORB_SLAM2/src/Optimizer.cc:1244:1:   required from here
/usr/include/eigen3/Eigen/src/Core/util/StaticAssert.h:32:40: error: static assertion failed: YOU_MIXED_DIFFERENT_NUMERIC_TYPES__YOU_NEED_TO_USE_THE_CAST_METHOD_OF_MATRIXBASE_TO_CAST_NUMERIC_TYPES_EXPLICITLY
     #define EIGEN_STATIC_ASSERT(X,MSG) static_assert(X,#MSG);
                                        ^
/usr/include/eigen3/Eigen/src/Core/util/XprHelper.h:707:3: note: in expansion of macro ‘EIGEN_STATIC_ASSERT’
   EIGEN_STATIC_ASSERT((internal::functor_is_product_like<BINOP>::ret \
   ^
/usr/include/eigen3/Eigen/src/Core/AssignEvaluator.h:745:3: note: in expansion of macro ‘EIGEN_CHECK_BINARY_COMPATIBILIY’
   EIGEN_CHECK_BINARY_COMPATIBILIY(Func,typename ActualDstTypeCleaned::Scalar,ty
   ^
CMakeFiles/ORB_SLAM2.dir/build.make:350: recipe for target 'CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o' failed
make[2]: *** [CMakeFiles/ORB_SLAM2.dir/src/Optimizer.cc.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/ORB_SLAM2.dir/all' failed
make[1]: *** [CMakeFiles/ORB_SLAM2.dir/all] Error 2
Makefile:83: recipe for target 'all' failed
make: *** [all] Error 2
a@ubuntu:~/ORB_SLAM2/build$

Matrixio, 2016-05-14:
my eigen version :
libeigen3-dev/xenial,xenial,xenial,xenial,now 3.3~beta1-2 all [installed]
  lightweight C++ template library for linear algebra

joyousrabbit, 2016-05-14:
@Matrixio 

Seems a bug. Could you try this fix:  https://github.com/ethz-asl/ethzasl_msf/pull/111/files   , as showing, replace all double_t with double.

liuyangyl, 2016-05-16:
I have met the same problem with @Matrixio , and I want to try the solution of @joyousrabbit , but I cannot find the files  msf_state_inl.h  and  msf_state.h . How could I find them ? Or maybe I did not have them?

sakurazhu, 2016-05-20:
yeh ,i can not find the  msf_state_inl.h and msf_state.h files either. 

2016-05-20 

myworldpig

amarant, 2016-05-20:
To correct it change [here](https://github.com/raulmur/ORB_SLAM2/blob/132d92fc86ced1926c00509337fd40b445d9e29c/Thirdparty/g2o/g2o/solvers/linear_solver_eigen.h#L56C70) `SparseMatrix::Index` to `int`. It's a dirty fix, bit it works.

Matrixio, 2016-05-25:
I solved the problem.
change "make -j" to "make -j1", 
my ubuntu run on a virtual machine.

best regards
Matrix

liuyangyl, 2016-05-26:
Thank you very much for the information. I will try that.

sakurazhu, 2016-05-27:
finally i have changed my ubuntu version to 14.04 exbot   and i  use   sudo apt-get install libeigen3-dev  to install eigen3  ,then there is no problem of eigen3 . i succeeded in doing ~ORB_SLAM2 ./build.sh 
but the finally step still have some problems. Anyway , thank you very much for all help .  @joyousrabbit @poine @Matrixio

ramtinkr, 2016-06-03:
Hello everyone,

I would like to run ORB-SLAM2 on Linux OS (Ubuntu 16.06 Xenial) with CPU intel Corei7 , 2.2 GHZ and I have tried to install the prerequisities, which were written in the README.md file on github.com. But the problem is I have faced several errors during building ORB_SLAM2 and compiling ORB_SLAM2  Examples for mono_tum.cc and mono_kitti.cc. It would be so kind if you can help me in this way because the type of error is somehow the same as above but the solution did not work for it.
[image]
[image]
[image]
[image]

amarant, 2016-06-03:
Have you tried the fix in my comment ?

ramtinkr, 2016-06-03:
@amarant Thanks for your response. Yes, I have replaced {sparseMatrix::Index} to {int} but the same error showed up. 
[image]

ramtinkr, 2016-06-03:
@amarant Thanks for your response. 
Yes, I have tried the fix by replacing {SparseMatrix::Index} to {int}. But the same error showed up.
[image]
[image]

liuyangyl, 2016-06-05:
Matrix solved the problem.
change "make -j" to "make -j1", 
his ubuntu run on a virtual machine.

--

Best regards,

刘洋 Liu Yang

中国科学院 国家空间科学中心
北京市海淀区中关村南二条一号
National Space Science Center, Chinese Academy of Sciences
No.1, Nan'ertiao Street, Zhongguancun, Haidian District, Beijing, China, 100190
Mob: +86-156-0060-2467
Email: liuyangyangyl@163.com

ramtinkr, 2016-06-05:
Thanks @liuyangyl. I have tried "make -j1" but the same error showed up.
[image]

sakurazhu, 2016-06-06:
maybe there is something wrong with your Pangolin .  when i met the problem i delected all the code about Openni* in Pangolin--src --CMakeLists  .  And then used make -j4 to make it successfully. Hope it works for you.

2016-06-06 

myworldpig

ramtinkr, 2016-06-06:
Thanks for your hint @sakurazhu . I have rebuilt my Pangolin successfully with make -j4 as you mentioned but unfortunately in rebuilding the ORB_SLAM2 I faced the same error. Today I am changing the version of my Ubuntu to 14.04 and try to rebuild ORB there.
[image]
[image]
[image]
[image]

ramtinkr, 2016-06-06:
Hi friends,

Today I changed the version of my Ubuntu to 14.04.04 LTS (Trusty Tahr) and I have built ORB-SLAM2 100 % successfully but at the end I got an error (CV_LOAD_IMAGE_UNCHANGED) of .CC Files of Examples { mono_tum.cc, mono_kitti.cc, rgbd_tum.cc, stereo_kitti.cc }. While compiling these file like mono_tum.cc, I also faced several errors related to CV_LOAD_IMAGE_UNCHANGED, Monotonic_clock, t1 & t2. Building Pangolin & Eigen were successful but after ORB_SLAM2 was build I found the built files locked, does anyone know how can I remove those errors and how to unlock the ORB "build" and "lib" Files? 

Thanks a lot

[image]
[image]

ramtinkr, 2016-06-07:
I solved the error of CV_LOAD_IMAGE_UNCHANGED by adding #include<opencv/highgui.h>. My only problem at the moment is "std::chrono::monotonic_clock", although I have the header #include <chrono>, it does not work.
[image]
[image]

shishiteng, 2016-08-17:
Thank you @amarant ,I modified "SparseMatrix::Index" to "int",now it works.

MartinXM, 2016-10-26:
I'm using Ubuntu 16.04 and met the same problem with @ramtinkr . The following trick works for me.
http://stackoverflow.com/questions/38647114/orb-slam-installation-on-ubuntu-xenial-16-04

xuyongzhi, 2016-10-27:
This works for me too. @Matrixio  Thank you! 
But I am still wandering are there any other problems while using ubuntu16.04 with ORB_SLAM2. For instance the ROS version has to be Kinetic instead of indigo. Do you know if ROS kinetic works for ORB_SLAM2?

cambazz, 2016-10-29:
@MartinXm how did you install the ubuntu 14 package in ubuntu 16?

cambazz, 2016-10-29:
The build was stopped at the same place, I tinkered for a while and installed libeigen2-dev and libeigen3-dev tru apt-get on ubuntu 16 xenial with roskinetic, and i believe i have a succesful build.

ros@daredevil:~/build/ORB_SLAM2$ ./build.sh 
Configuring and building Thirdparty/DBoW2 ...
mkdir: cannot create directory ‘build’: File exists
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ros/build/ORB_SLAM2/Thirdparty/DBoW2/build
[100%] Built target DBoW2
Configuring and building Thirdparty/g2o ...
mkdir: cannot create directory ‘build’: File exists
-- BUILD TYPE:Release
-- Compiling on Unix
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ros/build/ORB_SLAM2/Thirdparty/g2o/build
[100%] Built target g2o
Uncompress vocabulary ...
Configuring and building ORB_SLAM2 ...
mkdir: cannot create directory ‘build’: File exists
Build type: Release
-- Using flag -std=c++11.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/ros/build/ORB_SLAM2/build
[ 62%] Built target ORB_SLAM2
[ 68%] Built target mono_euroc
[ 75%] Built target mono_tum
[ 81%] Built target stereo_kitti
[ 87%] Built target mono_kitti
[ 93%] Built target stereo_euroc
[100%] Built target rgbd_tum