raulmur/ORB_SLAM2 #73: Error occur when building ROS Example

zsy372901, 2016-04-21:
Hi,
I have already build the thirdparty libraries and orb slam. I am trying follow section 7 to build the ROS Example. But when I execute command "make -j", I get the error:
[  0%] Built target rospack_genmsg_libexe
[  0%] Built target rosbuild_precompile
[ 50%] Building CXX object CMakeFiles/Mono.dir/src/ros_mono.cc.o
In file included from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/optimizable_graph.h:41:0,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/base_vertex.h:30,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/types_seven_dof_expmap.h:34,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/LoopClosing.h:34,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/LocalMapping.h:26,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/Tracking.h:31,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../include/System.h:29,
                 from /home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/src/ros_mono.cc:32:
/home/pkulab/ORB_SLAM2/Examples/ROS/ORB_SLAM2/../../../Thirdparty/g2o/g2o/types/../core/jacobian_workspace.h:30:22: fatal error: Eigen/Core: No such file or directory
 #include <Eigen/Core>
                      ^
compilation terminated.
make[2]: *** [CMakeFiles/Mono.dir/src/ros_mono.cc.o] Error 1
make[1]: *** [CMakeFiles/Mono.dir/all] Error 2
make: *** [all] Error 2
I do not know how to deal with it :(

romanomatthew23, 2016-04-22:
http://askubuntu.com/questions/491067/eigen-installation-seemed-to-work-but-i-still-cant-make-eigen-work

The comment on the first answer says 
"You also can make a link inside /usr/local/include to /usr/include/eigen3/Eigen so that you don't have to use any extra flag with g++ again. For doing that, just execute the following: sudo ln -s /usr/include/eigen3/Eigen /usr/local/include/Eigen"
And it worked for me!

xiesiyuan, 2016-12-31:
Worked for me ,thanks

SubMishMar, 2020-07-17:
> sudo ln -s /usr/include/eigen3/Eigen /usr/local/include/Eigen

Worked! Much thanks!