raulmur/ORB_SLAM2 #76: terminate called after throwing an instance of 'std::runtime_error'

DamonMIN, 2016-04-24:
Dear Sir:
In the running procedure,there is an error：
“terminate called after throwing an instance of 'std::runtime_error' what():  Pangolin X11: Unable to retrieve framebuffer options”.
I have tried many ways but failed. I need your generous help.
Thanks very much.

DamonMIN, 2016-04-24:
@raulmur Dear Sir:
I redo all the steps in another ubuntu system, but the error mentioned above is still existing. Please help me to solve it.

TheKiffer, 2016-04-26:
Having similar issues to this, other threads have commented that the problem exists due to a lack of video memory or the use of virtualisation software (VMWare...) 

https://github.com/stevenlovegrove/Pangolin/pull/81

Even after applying the fixes and testing on two seperate systems (GTX 650 1GB, GTX780 3GB), I am still getting the same error as you

DamonMIN, 2016-04-26:
I have solved the problems. Thank you for your clues. @TheKiffer

sakurazhu, 2016-05-27:
@TheKiffer have you solve the problem ?  i  have the same problem .

lucasmedino, 2016-07-12:
How did you solved the problem @DamonMIN

TheKiffer, 2016-07-13:
Are you running ORB-SLAM2 in VMWare @lucasmedino ? It seems the issue stems from running Ubuntu using virtualisation software since I have managed to get it up and running on a dedicated build without any issue. Here is how I solved it (the solution is provided at https://github.com/stevenlovegrove/Pangolin/pull/81) as the push which apparently fixed it didn't work for me. 

Find the device directory:
`cd Pangolin/src/display/device/
`

Open `display_x11.cpp`:
`gedit display_x11.cpp`

Comment out the following lines as such (Lines 123-124):
```
        //GLX_SAMPLE_BUFFERS  , glx_sample_buffers,
        //GLX_SAMPLES         , glx_sample_buffers > 0 ? glx_samples : 0,
```
Save the file, cd to the Pangolin build directory then rebuild:
```
cmake -DCPP11_NO_BOOST=1 ..
make -j
```

Try running a sequence afterwards and hopefully the runtime error has disappeared...hope this helps.