raulmur/ORB_SLAM2 #78: Running ORB-SLAM2 on a Mac

apatzer, 2016-04-26:
Just got ORB_SLAM2 up and running on a Mac.  This was a bit problematic, so I thought this might help some of you.

For those who are interested you may run into the following issues:
* You need to use GCC rather than the default CLang compiler. I did a `brew install gcc` then set these environment variables:
```
export CC=gcc-5
export CXX=g++-5
export CPP=cpp-5
export LD=gcc-5
```

* Can't find tr1/memory or tr1/unordered_map: This seems to be an issue with Boost `brew install boost --cc=gcc-5 --build-from-source` and was resolved by adding `-I/usr/local/Cellar/boost/1.60.0_2/include/boost/tr1/tr1` into the DBoW2 and g20 CMake files.

```
set(CMAKE_C_FLAGS "${CMAKE_C_FLAGS}  -Wall  -O3 -march=native -I/usr/local/Cellar/boost/1.60.0_2/include/boost/tr1/tr1")
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall  -O3 -march=native -I/usr/local/Cellar/boost/1.60.0_2/include/boost/tr1/tr1")
```

skylook, 2016-04-26:
https://github.com/skylook/ORB_SLAM2

poine, 2016-04-26:
@skylook: by the way, i am happy you used the binary dictionary. At lest
the evening i spent going through git tutorials was not totally in vain :)

apatzer, 2016-04-27:
@skylook Nice. But without Pangolin, how do you visualize what's going on? Speaking of which on a Mac Pangolin works fine alone, but in ORB-SLAM2 I keep getting CATransaction uncommitted, indicating the core animation thread is not working. Any suggestion to get around this AND have good visualization (eg Pangolin)?

skylook, 2016-04-27:
Thanks @poine your job really helps me :) 

@apatzer I remove Pangolin and other viewers because they are hard to port to other devices like iOS & Android. But if you just want to have a good visualization on Mac you can use OpenCV to project points like the code in Examples/Monocular/Main.cc.

There is also a good example of shomin & poine (https://github.com/shomin/ORB_SLAM2) which sepreate Pangolin and viewers may help (but they are not targeted on Mac).

apatzer, 2016-04-27:
@skylook Your visualization using OpenCV works perfectly for me, and does not seem to incur any real performance hit for doing the visualization. Thanks!

TimingSpace, 2016-05-20:
@apatzer @skylook Hi, could you please tell me where is the file named "Examples/Monocular/Main.cc"?

skylook, 2016-05-20:
It's in my git here, you can find it here:
https://github.com/skylook/ORB_SLAM2/blob/master/Examples/Monocular/Main.cc

TimingSpace, 2016-05-20:
@skylook  Thank you very much! It works!

bikong2, 2016-09-01:
@apatzer i get the same problem with you. i compiled ORB_SLAM2 on mac, and can running successfully when i disable Map and View drawing. but it failed when i enable them. It sames that problem occur because of multi-thread problem, do you have any solutions here?

hoverpeter, 2017-01-17:
@apatzer The issue happened to me as well. The main cause for that is viewer thread has to be in the main thread in this case. I don't know why. However, I practically work around it by knowing this.

Ethan-Zhou, 2017-06-21:
@hoverpeter Hey, any progress on this pangolin issue? I came across the same problem. Have you figured out how to make the original visualization work without putting the viewer into the main thread?

pixelcowboy, 2017-06-27:
@Ethan-Zhou - any luck? Are you able to get working?

@hoverpeter - Can you post your modifications (code sample) for your work around? 

Thanks in advance.

kitizz, 2017-08-10:
A little late to the game, but I've modified my fork to be (I think) out of the box compatible with Mac. Hasn't been extensively tested, see how you go.
https://github.com/kitizz/ORB_SLAM2

pixelcowboy, 2017-08-15:
I've downloaded and tested @kitizz fork. Works good for me on MacOS. Thanks!

Two comments:

1. Cmd line output is "noisy" with "Frame" counter and "CreateNewMapPoints" echoing. Perhaps a debug remnant?

2. Documentation READE.md should be updated according to MacOS support.

Very nice! Again, thanks!

kitizz, 2017-08-26:
Woops! Damn. Right you are. Should be fixed up now!
I've also clarified that I've only modified the `mono_tum` executable to get Pangolin working on Mac. The same changes would have to be carried over to any other executable as needed.

Glad it was useful :)

gentlemanman, 2018-01-18:
@apatzer what's the meaning of last two line

gentlemanman, 2018-01-18:
@apatzer the boost is 1.66.0 , I can't find "boost/1.60.0_2/include/boost/tr1/tr1".  In a word , 1.66.0 don't have"/tr1/tr1".

IvyGongoogle, 2019-08-19:
kitizz‘s repo works. @kitizz  thanks~

mohGhazala96, 2020-09-24:
> Anyone else having trouble w/ @kitizz repo?
> Running macOS 10.14.6 and getting following error (start of stack trace) when building orb slam:
> 
> ```
> In file included from orb_slam_mac/src/LoopClosing.cc:21:
> In file included from orb_slam_mac/include/LoopClosing.h:24:
> In file included from orb_slam_mac/include/KeyFrame.h:24:
> In file included from orb_slam_mac/include/MapPoint.h:25:
> In file included from /orb_slam_mac/include/Frame.h:27:
> In file included from orb_slam_mac/Thirdparty/DBoW2/DBoW2/BowVector.h:14:
> /Library/Developer/CommandLineTools/usr/include/c++/v1/map:886:5: error: static_assert failed due to requirement 'is_same<typename allocator_type::value_type, value_type>::value' "Allocator::value_type must be same type as value_type"
>     static_assert((is_same<typename allocator_type::value_type, value_type>::value),
>     ^              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> orb_slam_mac/src/LoopClosing.cc:439:21: note: in instantiation of template class 'std::__1::map<ORB_SLAM2::KeyFrame *, g2o::Sim3, std::__1::less<ORB_SLAM2::KeyFrame *>,
>       Eigen::aligned_allocator<std::__1::pair<const ORB_SLAM2::KeyFrame *, g2o::Sim3> > >' requested here
>     KeyFrameAndPose CorrectedSim3, NonCorrectedSim3;
>                     ^
> orb_slam_mac/src/LoopClosing.cc:469:29: error: no viable overloaded operator[] for type 'ORB_SLAM2::LoopClosing::KeyFrameAndPose' (aka 'map<ORB_SLAM2::KeyFrame *, g2o::Sim3, std::less<KeyFrame *>,
>       Eigen::aligned_allocator<std::pair<const KeyFrame *, g2o::Sim3> > >')
>             NonCorrectedSim3[pKFi]=g2oSiw;
>             ~~~~~~~~~~~~~~~~^~~~~
> In file included from orb_slam_mac/src/Optimizer.cc:23:
> In file included from orb_slam_mac/Thirdparty/g2o/g2o/core/block_solver.h:30:
> In file included from orb_slam_mac/Thirdparty/g2o/g2o/types/../core/solver.h:32:
> In file included from orb_slam_mac/Thirdparty/g2o/g2o/types/../core/sparse_block_matrix.h:229:
> orb_slam_mac/Thirdparty/g2o/g2o/types/../core/sparse_block_matrix.hpp:339:45: warning: ISO C++ specifies that qualified reference to 'SparseBlockMatrix' is a constructor name rather than a type in
>       this context, despite preceding 'typename' keyword [-Winjected-class-name]
>     typename SparseBlockMatrix<MatrixType>::SparseBlockMatrix* s=new SparseBlockMatrix(rowIdx, colIdx, m, n, true);
>                                             ^
> orb_slam_mac/src/LoopClosing.cc:473:39: error: use of undeclared identifier 'mit'; did you mean 'min'?
>         for(KeyFrameAndPose::iterator mit=CorrectedSim3.begin(), mend=CorrectedSim3.end(); mit!=mend; mit++)
> ```

I am too, has anyone solved it?

muzizhi, 2020-11-18:
> Anyone else having trouble w/ @kitizz repo?
> Running macOS 10.14.6 and getting following error (start of stack trace) when building orb slam:
>
> ```
> In file included from [...]

that seems like I meet the same error with AnselmC. I use macOS 10.13.6 by the way.
so anyone solve this problem?

muzizhi, 2020-11-26:
> I am too, has anyone solved it?

anyone solved it???