raulmur/ORB_SLAM2 #79: Monocular TUM example runs single thread on a single CPU

apatzer, 2016-04-26:
Just got ORB-SLAM2 up and running on a Mac. When I run the Monocular TUM on the Frieburg1 dataset, it runs for hours (still going).

```
./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ~/Downloads/rgbd_dataset_freiburg1_xyz/rgb
```

Activity Monitor shows a single CPU (I have 8 cores...7 of them not being used) and a single thread. Where's the real-time performance? Have I compiled things incorrectly? I'm on a 16GB machine with a solid processor, but no CUDA (Mac's use an Intel GPU).

[image]

TheKiffer, 2016-04-26:
Change the command for it to point at the sequence folder and not the image sequence within:

`./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUM1.yaml ~/Downloads/rgbd_dataset_freiburg1_xyz/`

apatzer, 2016-04-27:
Yep, that works. Otherwise it's caught in an infinite loop.

closed 2016-04-27 by apatzer