raulmur/ORB_SLAM2 #80: Left-handed coordinate frame?

jacobperron, 2016-04-27:
Hi, just want to make sure I am using ORB-SLAM's camera poses correctly. From my tests, it appears the system uses a "left-handed" coordinate frame:

+x is to the left
+y is up
+z is backwards

Can someone else confirm this is true?
Thanks.

poine, 2016-04-27:
@Jacob: not sure how to interpret left up backward, but when i compute a
transformation for georeferencing (ECEF) the slam output, i obtain a
negative scale factor, which would tend to make me think that there is
something left handed in ORBSLAM2.

yeduvaka, 2017-07-10:
Hey @poine , I was trying to convert SLAM co-ordinates to GPS/Some global co-ordinate system, however I have been getting not very accurate results i.e: a lot of the time the scale is off. Can you tell if and how you got accurate georeferencing?

floatlazer, 2018-05-23:
Based on what I observed, I think ORB-SLAM uses a right handed coordinate system with
+x to right
+y to down
+z to forward

mmahdavian, 2020-09-25:
@jacobperron The orb-slam is right handed. But I am guessing you are checking projection matrix last column to find the camera position and that becomes left handed. Because the last column is basically -Rc. c is camera pose wrt world. But the - makes it world frame wrt camera and it is not correct to use that to check the camera position and that is left handed.You need to negate all values of the last column to find camera position wrt world and that is right handed.

jacobperron, 2020-09-28:
@mmahdavian Thanks for response. I was probably looking at the projection matrix. I'll close this ticket since it was more of a question than an actual issue.

closed 2020-09-28 by jacobperron