raulmur/ORB_SLAM2 #81: Cannot view Pangolin in mono_tum

apatzer, 2016-04-27:
I'm trying to run a Monocular TUM example. The main thread seems to execute just fine, but I get a warning that there is a Core Animation Transaction (CATransaction) that is not being processed, originating from `PangolinAppDelegate window`

This could be a Mac specific issue.  Any idea how to solve this?

I can run all of the Pangolin examples {SimpleDisplay, SimplePlot, SimpleVideo, MultiDisplay, etc} and they pop up just fine on the Mac. I'm simply not able to visualize inside of ORB-SLAM2.

```
CoreAnimation: warning, deleted thread with uncommitted CATransaction; created by:
0   QuartzCore                          0x00007fff8dda569a _ZN2CA11Transaction4pushEv + 318
1   QuartzCore                          0x00007fff8dda519a _ZN2CA11Transaction15ensure_implicitEv + 276
2   QuartzCore                          0x00007fff8ddaeaa6 _ZN2CA11Transaction8set_seedEj + 30
3   QuartzCore                          0x00007fff8ddaea60 +[CATransaction(CATransactionPrivate) generateSeed] + 56
4   AppKit                              0x00007fff8436ea51 +[NSDisplayCycle currentDisplayCycle] + 37
5   AppKit                              0x00007fff8436e9a3 -[NSDisplayCycleObserver setNeedsUpdateStructuralRegions:] + 34
6   AppKit                              0x00007fff8437bf76 -[NSThemeFrame _tileTitlebarAndRedisplay:] + 123
7   AppKit                              0x00007fff84370bc3 -[NSThemeFrame _updateButtons] + 3068
8   AppKit                              0x00007fff84368a26 -[NSThemeFrame _floatTitlebarAndToolbarFromInit:] + 161
9   AppKit                              0x00007fff84365f6d -[NSThemeFrame initWithFrame:styleMask:owner:] + 243
10  AppKit                              0x00007fff84364a7c -[NSWindow _commonInitFrame:styleMask:backing:defer:] + 568
11  AppKit                              0x00007fff843630c7 -[NSWindow _initContent:styleMask:backing:defer:contentView:] + 1475
12  AppKit                              0x00007fff84362afd -[NSWindow initWithContentRect:styleMask:backing:defer:] + 45
13  libpangolin.dylib                   0x0000000106f08aa9 -[PangolinAppDelegate window] + 201
14  libpangolin.dylib                   0x0000000106f08e58 _ZN8pangolin19CreateWindowAndBindENSt3__112basic_stringIcNS0_11char_traitsIcEENS0_9allocatorIcEEEEiiRKNS_6ParamsE + 248
15  libORB_SLAM2.dylib                  0x0000000104883b64 _ZN9ORB_SLAM26Viewer3RunEv + 116
```

On a (possibly) related topic, I get a `seg 11` at the end of running mono_tum.  It makes it past saving the trajectory, and when I step through the code in GDB, thread 0 makes it to exit(), so I would guess this is happening due to activity on another thread?

```
median tracking time: 0.0260916
mean tracking time: 0.029683

Saving keyframe trajectory to KeyFrameTrajectory.txt ...

trajectory saved!
Segmentation fault: 11
```

apatzer, 2016-04-27:
One more oddity: a window pops up, but shuts down before I can really see anything. I assume that this is the Pangolin Viewer, but it's hard to tell as it is literally just a flash.

apatzer, 2016-04-27:
@skylook Did you ever get Pangolin to run with ORB-SLAM2 on a Mac?  Pangolin runs just fine on it's own, but the rendering thread seems to die within ORB-SLAM2.

coskunyerli, 2016-04-28:
I get an error but not like this
It is "Crashed Thread:        5  Dispatch queue: com.apple.Xplugin.CGEventPort"
I built pangolin with x11 library 

also I build pangolin with cocoa framework again it did not work 
I got error with cocoa framework "libc++abi.dylib NS Exeption error"

skylook, 2016-04-29:
Sorry @apatzer , I haven't tried Pangolin on Mac.

mhsung, 2016-06-21:
I got the same with @apatzer (Segmentation fault: 11). Any solution for that?

bikong2, 2016-08-30:
i got the same error with @apatzer .
Any solution someone would have found?
many thanks if there is a solution.

CodeGenerator, 2016-10-17:
All problems here appear because System tries to execute `Viewer::Run` not on main thread. Working with UI libraries not on main thread is error prone.

fschopp, 2016-10-20:
You may want to try https://github.com/raulmur/ORB_SLAM2/pull/179