raulmur/ORB_SLAM2 #82: Is there any way to reduce memory usage?

skylook, 2016-04-27:
In my test, there is about 500MB memory occupied by ORB SLAM.  That seems to be unacceptable for mobile phones.
Most of the usage comes from BOW vocabulary and keyframe database I suppose.
Is there any way to reduce the memory usage?

raulmur, 2016-04-27:
You can create your own vocabulary using the "create" function in Thirdparty/DBoW2/DBoW2/TemplatedVocabulary.h  and then save it. Using a vocabulary of 5 levels instead of 6 will reduce in a factor of 10 the memory without a significant lost in performance. We give some details on how we created the default vocabulary here: http://webdiis.unizar.es/~jdtardos/papers/2014_IEEE_ICRA_Mur_Tardos.pdf

weblucas, 2016-04-27:
Hi Raul,

Where can I  download the Bovisa dataset?

Thank you

Lucas

raulmur, 2016-04-27:
http://www.rawseeds.org/rs/datasets/view//7

weblucas, 2016-04-27:
I tried these torrents, but nobody is sharing this torrents. Do you know
where else I can try to donwload this?

raulmur, 2016-04-27:
I did not know this was not working... 
Anyway you can use other sequences. For example you could mix the outdoors NewCollege (http://www.robots.ox.ac.uk/NewCollegeData/) and the indoors MIT Stata Center (http://projects.csail.mit.edu/stata/downloads.php)

skylook, 2016-04-29:
Thanks, that's really helps, I will try that :)

closed 2016-04-29 by skylook

realitian, 2017-08-07:
Can anybody help me by sharing the recreated vocabulary of 5 levels instead of 6?