raulmur/ORB_SLAM2 #84: Crash in RGBD + ROS

marcelinomalmeidan, 2016-04-29:
Hello Everyone,

I was testing ORB_SLAM2 with my new Intel RealSense R200, but I am having some trouble with it.

If I run in Monocular mode, I don't have much trouble. It runs and tracks whenever it feels like it initialized well.

However, when I go to RGBD mode, the Vocabulary file downloads, the program starts (the screens popup), and crashes immediately after. This is what shows-up in the whole procedure:

`Input sensor was set to: RGB-D

Loading ORB Vocabulary. This could take a while...
Vocabulary loaded!

Camera Parameters: 
- fx: 621.437
- fy: 619.083
- cx: 323.015
- cy: 233.425
- k1: 0.167161
- k2: -0.488226
- p1: 0.001355
- p2: 0.00031
- fps: 60
- color order: RGB (ignored if grayscale)

ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

Depth Threshold (Close/Far Points): 3.21835
New map created with 305 points
[ORB_SLAM2-1] process has died [pid 6971, exit code -11, cmd /home/mma2739/ORB_SLAM2/Examples/ROS/ORB_SLAM2/RGBD /home/mma2739/ORB_SLAM2/Vocabulary/ORBvoc.txt /home/mma2739/Desktop/Marcelino/RealSense.yaml /camera/rgb/image_raw:=/camera/color/image_raw /camera/depth_registered/image_raw:=/camera/depth/image_raw __name:=ORB_SLAM2 __log:=/home/mma2739/.ros/log/6b36e536-0dac-11e6-bdbb-0c8bfdd15353/ORB_SLAM2-1.log].
log file: /home/mma2739/.ros/log/6b36e536-0dac-11e6-bdbb-0c8bfdd15353/ORB_SLAM2-1*.log
all processes on machine have died, roslaunch will exit`

Does anyone have any guess for why this is happening? I've been trying to understand this for hours now and I don't know where to start.

FPSychotic, 2019-01-18:
can you help me to make it work at least in monocular please

FPSychotic, 2019-01-19:
Ros dont suport depth HW registered, must be in Sw in the launch rgb or maybe you are suscribed to that topic,, maybe is that your issue