raulmur/ORB_SLAM2 #92: Disable viewer

durn2000, 2016-05-09:
Is there a way to disable the viewer from running in ORB SLAM2? I am running the process through ssh and it crashes when it tries to open the viewer display.

poine, 2016-05-09:
AFAIK there is nothing like this in the official codebase. I have done
something like this for my personal use:
http://recherche.enac.fr/~drouin/slam/orbslam2/poine_orbslam2_04_07_16.tgz

durn2000, 2016-05-09:
Actually, after I dug into it a little further I found that the System class (System.cc) has a boolean variable called bUseViewer that can be used to enable/disable the viewer. I guess I should have dug a little deeper before posting.

ank700, 2016-07-27:
Hi @durn2000 , I set the variable bUseViewer to false but even then the map viewer pops up. What else did you change?
Thanks

shadowver, 2017-04-24:
There is an option in System.h and System.cc that allows users to turn off the Viewer in System(constructor). However, I have tried couple of times and it seems not very stable. Out of 10 times I had tired, six of them caused tracking lost and returned nothing but force window closure. Anyone know what may cause such issue?

chwimmer, 2017-04-24:
Tracking Lost has nothing to do with the viewer.

There are or were issues when you don't start the viewer and the System tries to reset. But this issues should be fixed now in the current version of ORBSLAM2

shadowver, 2017-04-24:
Thanks @chwimmer for the reply. I downloaded the software like four days ago, I assumed that is the latest version? The issue is pretty odd as long as I turn on the viewer, everything goes perfectly. 

p.s. for testing, I did not move my Depth Sensor at all.

chwimmer, 2017-04-24:
That's really pretty odd because I just can repeat it that the Tracking Lost has nothing to do with the viewer. That the program crashes when the viewer is turned of when the system tries to reset. But when you cloned the repository about 4 days ago it should be the stable version. Nevertheless can you check if [this commit](https://github.com/raulmur/ORB_SLAM2/commit/2f1b9b0e15f31c06789befb126d9ba8f6c93de46)  is in your current version? 

Do you have any output you get from ORBSLAM2?

shadowver, 2017-04-24:
I think so too. I mean the the viewer is really just depends on the GL part, it should not affect the tracking function. Rarely, I can extract the pose matrix when the Viewer is off, but most of the time the program just stuck on the system resetting phrase. I will download repository again and try. Will let you guys know what went wrong if I find the problem.

chwimmer, 2017-04-24:
Ah ok that sounds like the ORBSLAM2 cant extract enough features. How does you test scenario look like? Have you enough features that ORBSLAM2 can extract? How does you depthimages look like? Have you enough valid depth values?

shadowver, 2017-04-24:
exact the same code and the same camera position(facing static direction), viewer on works great, but did not work while viewer is off.

hoangthien94, 2017-11-05:
If I disable UI of mono_tum.cc it gets stuck (hang) forever after this:

`ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7`
Has anyone come up with a workaround for this?

shadowver, 2017-11-06:
@hoangthien94 what is the error message?

hoangthien94, 2017-11-06:
With **bUseViewer = true** I get:
```
ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7

-------
Start processing sequence ...
Images in the sequence: 798

New Map created with 82 points
```

With **bUseViewer = false** I get:
```
ORB Extractor Parameters: 
- Number of Features: 1000
- Scale Levels: 8
- Scale Factor: 1.2
- Initial Fast Threshold: 20
- Minimum Fast Threshold: 7
```
No more output after that (I waited for about 1 hour)

shadowver, 2017-11-06:
@hoangthien94  It looks like the initialization is wrong. I have not tried with TUM dataset, but if you take a look at the tracking.cc file (line 675), trackedMappoints has to larger than 100 points in order to initialize the mono slam. You can tried to reduce the parameter to say 80 according to your error message and rebuild the program to see whether it works or not.