raulmur/ORB_SLAM2 #94: Build ORB_SLAM2 on OSX El Captain

Gaoxiang-Zhang, 2016-05-17:
I was following the instructions given by [Running ORB-SLAM2 on a Mac](https://github.com/raulmur/ORB_SLAM2/issues/78)
But I met with the following error:
```
make[2]: *** No rule to make target `../Thirdparty/DBoW2/lib/libDBoW2.so', needed by `../lib/libORB_SLAM2.dylib'.  Stop.
make[2]: *** Waiting for unfinished jobs....
```
I found libDBoW2.dylib but not .so file in the Thirdparty/DBoW2/lib/ folder.
Is there something wrong with my configuration? I am looking forward for an answer.

coskunyerli, 2016-05-19:
You must change CMakeList file.

in CMakeList, there is a line like Thirdparty/DBoW2/lib/libDBoW2.so

change into libDBow.dylib and libg2o.dylib

but in osx, orbslam is not built you must use linux to use orbslam. if you run orbslam, please contact me.

closed 2016-05-23 by Gaoxiang-Zhang

cduguet, 2016-08-05:
I just compiled orbslam in OSX, I have not attempted to run it yet (compiling more stuff right now), but will try to remember to let you know if it works (or doesn't).

elliotwoods, 2017-02-14:
would be great to hear more about this. looking for windows and osx builds to try out
currently attempting both and will upload a fork if i succeed at either