raulmur/ORB_SLAM2 #95: Why quaternion domain are 120 to -240 degree?

tdnet12434, 2016-05-21:
I published orientation from ros_mono.cc

```
        cv::Mat Rwc = mTcw.rowRange(0,3).colRange(0,3).t();
        cv::Mat twc = -Rwc*mTcw.rowRange(0,3).col(3);
        tf::Matrix3x3 M(Rwc.at<float>(0,0),Rwc.at<float>(0,1),Rwc.at<float>(0,2),
                        Rwc.at<float>(1,0),Rwc.at<float>(1,1),Rwc.at<float>(1,2),
                        Rwc.at<float>(2,0),Rwc.at<float>(2,1),Rwc.at<float>(2,2));
        tf::Vector3 V(twc.at<float>(0), twc.at<float>(1), twc.at<float>(2));

        tf::Transform tfTcw(M,V);

        tf::Quaternion Q = tfTcw.getRotation();
        
        export_odom_data.pose.pose.position.x = V.x();
        export_odom_data.pose.pose.position.y = V.y();
        export_odom_data.pose.pose.position.z = V.z();
        export_odom_data.pose.pose.orientation.x = Q.getX();
        export_odom_data.pose.pose.orientation.y = Q.getY();
        export_odom_data.pose.pose.orientation.z = Q.getZ();
        export_odom_data.pose.pose.orientation.w = Q.getW();
```

When i try to plot graph between SLAM and IMU 
it look like not the same domain that it should be.

How can i manually change domain?

[image]

(iw , iz  are imu orientation w, z )

ank700, 2016-06-02:
Hi @tdnet12434 , I have a few questions. 
1) How are you making Mat = mpslam in the first line. When I do that, it gives the following error:

OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file /home/ros/sources/opencv/opencv-2.4.9/modules/core/src/matrix.cpp, line 269
terminate called after throwing an instance of 'cv::Exception'
  what():  /home/ros/sources/opencv/opencv-2.4.9/modules/core/src/matrix.cpp:269: error: (-215) m.dims >= 2 in function Mat

Aborted (core dumped)

2) Is the pose with respect to world? Or is it the pose of only a keyframe in reference to the previous keyframe?

Thanks

Thomas00010111, 2016-06-03:
I wrote a ros node for rgbd. Maybe that helps
https://github.com/raulmur/ORB_SLAM2/pull/102/commits/5ec159225510504603ef8aee3a7732559e0c6999

camera uses left handed coordinate system.
Did you transform to world coordinates?

tdnet12434, 2016-06-04:
Thank you very much !! @Thomas00010111 
I got your idea then i apply and rebuild it to ENU coordinate.
So, it's now look like this
`````
//construct form q(x,y,z,w)

  tf::Quaternion q_vc_rebuild(-q_vc.z(),
                               q_vc.x(),
                               q_vc.y(),
                              -q_vc.w());
```

closed 2016-06-15 by tdnet12434

ank700, 2016-06-17:
Hello @tdnet12434 , I am getting correct results following your code but I have a doubt regarding your calculations in the first comment.
You are calculating twc as follows: 

cv::Mat twc = -Rwc*mTcw.rowRange(0,3).col(3);

How is this possible? Generally, the transformation matrix (mTcw) consists of a 3x3 rotation matrix and 3x1 translation matrix. Based on this the 4th column of mTcw should siply give the translation. Can you please explain why is it different here?

Thanks

tdnet12434, 2016-06-17:
@ank700 
i just copy and paste from ORB_SLAM first version that written by author, so i don't sure my understanding is correct. :D 

i think, when we slice mTcw for translation, it's translation "world respect by camera frame".
so it need to transform to "camera respect to world frame" by Rwc.

ank700, 2016-06-23:
Hi, I have one more problem with transforms- hope you can help. I did the transforms as follows:

//Apply Transform to get correct pose values in world frame
    Rwc = pose.rowRange(0,3).colRange(0,3).t();
    twc = -Rwc * pose.rowRange(0,3).col(3);

    tf::Matrix3x3 M(-Rwc.at<float>(0,0),Rwc.at<float>(0,1),Rwc.at<float>(0,2),
                    -Rwc.at<float>(1,0),Rwc.at<float>(1,1),Rwc.at<float>(1,2),
                    Rwc.at<float>(2,0),-Rwc.at<float>(2,1),-Rwc.at<float>(2,2));
    tf::Vector3 V(twc.at<float>(0), twc.at<float>(1), twc.at<float>(2));

Then, I rotate the above transformation by 180 degree about y axis to get the following camera coordinate frames - +z forward, +x right, +y down.

        trans_rot.setOrigin(tf::Vector3(0,0,0));
        trans_rot.setRotation(tf::Quaternion(0, 1, 0, 0));  

        tf::Transform tfTcw = trans1 * trans_rot;

[image]

This is how the camera frame looks like. 
Now the problem is that the rotation about a single axis is inverted (in this case y and x). Can you please tell me what should I change in the rotation matrix/quaternion to invert the rotation?

Thanks

ramtinkr, 2016-08-18:
@tdnet12434 Hi, I'd like to publish Camera Poses in ORB_SLAM2. From the command lines that you have written, I could not understand exactly which part of the ros_mono.cc I should add them, but I need exactly this pose information (3D Pose+4D Orientation) in real-time. It would be great if you can give me some guidlines, that how can I add this option to publish camera poses in ORB_SLAM2 by myself.

       
        export_odom_data.pose.pose.position.x = V.x();
        export_odom_data.pose.pose.position.y = V.y();
        export_odom_data.pose.pose.position.z = V.z();
        export_odom_data.pose.pose.orientation.x = Q.getX();
        export_odom_data.pose.pose.orientation.y = Q.getY();
        export_odom_data.pose.pose.orientation.z = Q.getZ();
        export_odom_data.pose.pose.orientation.w = Q.getW();

I have already done that for LSD-SLAM, which had the topic Camera/pose.

Thanks a lot
Ramtin

tdnet12434, 2016-08-18:
@ramtinkr Hi, i pushed that my topic code lines in ImageGrabber::GrabImage because TrackMonkcular function returned pose matrix already.

The output of that matrix will be camera frame coordinate. So we need to convert some axis and direction to be ENU coordinate following second commend (that i do after receive this topic in other node).