raulmur/ORB_SLAM2 #97: Segmentation fault (core dumped) when running custom datasets

Gaoxiang-Zhang, 2016-05-23:
The kitti and tum work well, but my custom example with almost the same code throws (because of empty `imread` result):
```
Failed to load image at: /home/user/vehicle/nexus/16_16_5/rgb/1463385917621.jpg
Segmentation fault (core dumped)
```
I tried to write a small program simply calling `imread` function with the same path, there was nothing wrong. 
And I tried to replace the image file with png format, the error still exists.
It is the fault caught in image reading process, so I think the wrong parameters in my yaml file or irregular image timestamp are not related.
It's weird and I am looking forward to an answer. Thank you!

halajun, 2017-07-26:
@Gaoxiang-Zhang  I also have this strange problem, have you solved it?

yingyingqin, 2018-03-13:
I also meet the problem, Have you solved it? I can run finished one dataset, but can't run anonther, I don't know why. @Gaoxiang-Zhang @halajun