raulmur/ORB_SLAM2 #100: [bug] crash when pRefKF->mTcwBefGBA is NULL

skylook, 2016-05-25:
I met a crash with LoopClosing::RunGlobalBundleAdjustment function in this line:
> cv::Mat Rcw = pRefKF->mTcwBefGBA.rowRange(0,3).colRange(0,3);
pRefKF->mTcwBefGBA is NULL, but I cannot reproduce it every time.

riematrix, 2016-06-28:
@skylook Hi, I have the crash error after global bundle adjustment. I'm not sure if it is the same with yours. My error stack is: 
"terminate called after throwing an instance of 'cv::Exception'
  what():  /build/buildd/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp:269: error: (-215) m.dims >= 2 in function Mat"
Do you have the same error?

skylook, 2016-07-04:
Yes, it seems to be the same :)

YonatanSimson, 2016-07-11:
Had the same error on one of the Euroc MAV examples

YAZAN-MAHER, 2019-10-04:
I have the same problem :
`OpenCV Error: Assertion failed (m.dims >= 2) in Mat, file /build/opencv-_msWgW/opencv-2.4.8+dfsg1/modules/core/src/matrix.cpp, line 269`