raulmur/ORB_SLAM2 #101: Integrate GPS to ORB_SLAM2

yilun, 2016-05-25:
Hi @raulmur and all forks, Im trying to integrate GPS data into the ORB_SLAM2 to correct the scale drift. 

I had tried to put GPS translation into tracking and localmapping thread by doing a light BA when a new keyframe is inserted, but BA failed when the new translations. Im now trying to put it into loopclosing thread using SIM3 and global BA, because I think correcting drift is similar to closing the loop, but not working yet. 

So, could you give some advice?

Regards

AlejandroSilvestri, 2017-03-25:
@yilun , it isn't an easy task.

If you correct actual keyframe pose with GPS info, you should update all map with the same procedure used to close loops, evenly distributing the error in all poses.

s894330, 2017-12-22:
Hi @raulmur 

The stereo kitti dataset has GPS data for each frame image，I want to use GPS pose to set each frame's pose to help map accuracy (I using kitty dataset 08 for test, because dataset 08 has no loop closure, cause the map has big gap with true map scale), I set the frame pose in "StereoInitializaiton" for first frame, "TrackReferenceKeyFrame" for second frame, and "TrackWithMotionModel" for other frames, but this will happen tracking lost soon after

Currently I have no idea about how to pus GPS data into ORB SLAM to help map accuracy, could you give some advice? 

thank you :)

Teemoshan, 2021-05-06:
@yilun Hi yilun
I am trying to integrate GPS work, I note that you put GPS into loopclosing thread using SIM3 and global BA, Have you succeeded？How much has the effect improved？
Thank you