raulmur/ORB_SLAM2 #103: OpenCV error when running ros examples  in Ubuntu16.04 + ROS (kinetic version)

gblack007, 2016-05-31:
Hi,
I tried to run the ros examples in Ubuntu16.04 + ROS (kinetic version), run the following command at each tab:
  roscore
  rosrun ORB_SLAM2 Stereo ORB_SLAM2/Vocabulary/ORBvoc.txt ORB_SLAM2/Examples/Stereo/EuRoC.yaml true
  rosbag play --pause Downloads/MH_01_easy.bag  /cam0/image_raw:=/camera/left/image_raw /cam1/image_raw:=/camera/right/image_raw
 I got this problems:
OpenCV Error: Bad argument (Invalid pointer to file storage) in cvGetFileNodeByName, file /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/core/src/persistence.cpp, line 709
terminate called after throwing an instance of 'cv::Exception'
  what():  /tmp/binarydeb/ros-kinetic-opencv3-3.1.0/modules/core/src/persistence.cpp:709: error: (-5) Invalid pointer to file storage in function cvGetFileNodeByName..

How to solve the problem?
Hope to you reply, thank u very much.

StephMc, 2016-06-18:
I have the same problem (using ros-indigo-opencv3-3.1.0), @gblack007 did you manage to solve it?

emphos1, 2016-07-31:
any idea?

evclark, 2016-08-08:
Me too, on ros kinetic on Ubuntu 16.04, running ```rosrun ORB_SLAM2 Mono Vocabulary/ORBvoc.txt Examples/Monocular/TUM3.yaml ```. Any ideas?

poine, 2016-08-21:
This is likely an opencv2 vs opencv3 issue. I have built ORBSLAM2 on Ubuntu
16.04 with ROS kinetic this weekend and have had to change the
CMakeLists.txt and all the include opencv occurrences to get it working.
Please try ldd on lib/libORB_SLAM2.so

I0x0I, 2016-08-26:
I solve the problem by running `apt-get autoremove libopencv2.4` in Ubuntu 16.04 as mentioned in [this page](https://github.com/alejocb/dpptam/issues/5)

jiawei-mo, 2016-10-04:
It uses ros-kinetic-opencv3-3.1.0 already, so removing libopencv outside of ROS does not help.

chiragmajithia, 2016-10-14:
Is the issue resolved? Facing same problem.
Did autoremove libopencv2.4 worked?

jiawei-mo, 2016-10-14:
I went back to 14.04 and everything is fine now.

chiragmajithia, 2016-10-18:
@poine Hey, could you please share the files with changes with opencv3 / opencv2 conflict? I do not want to reinstall 14.04 again.
Thanks

fanyuzeng, 2016-11-01:
CMakeLists.txt 
find_package(OpenCV xxx REQUIRED)
The version xxx should be consistent with the version of installed opencv

geokarant, 2016-11-01:
in this part i just used "find_package(OpenCV REQUIRED) " and i didnt
modified it.

cmxnono, 2016-11-14:
Because ROS kinetic is depends on opencv3.1.0, and you can find libopencv_*3.so at /opt/ros/kinetic/lib;

artifactz, 2016-12-07:
I have
* changed `find_package(OpenCV 2.4.3 REQUIRED)` to `find_package(OpenCV 3.1.0 REQUIRED)`
* `apt autoremove libopencv2.4` (I cannot `sudo apt remove libopencv-*`, because ROS depends on it.)
* rebuilt ORB_SLAM2 with `cmake .. -DCMAKE_BUILD_TYPE=Release -DOpenCV_DIR=/opt/ros/kinetic/; make`

But ORB_SLAM2 crashes with a segmentation fault in

    (gdb) where
    #0  0x0000000000000000 in ?? ()
    #1  0x00007ffff0515616 in void cv::FAST_t<16>(cv::_InputArray const&, std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >&, int, bool) ()
       from /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4
    #2  0x00007ffff5e06401 in ORB_SLAM2::ORBextractor::ComputeKeyPointsOctTree(std::vector<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> >, std::allocator<std::vector<cv::KeyPoint, std::allocator<cv::KeyPoint> > > >&) ()
    ...

indicating, that it still tries to use some OpenCV 2.4 libs. Even if

    ldd libORB_SLAM2.so | grep cv
    	libopencv_calib3d3.so.3.1 => /opt/ros/kinetic/lib/libopencv_calib3d3.so.3.1 (0x00007f2423d25000)
    	libopencv_features2d3.so.3.1 => /opt/ros/kinetic/lib/libopencv_features2d3.so.3.1 (0x00007f2423a63000)
    	libopencv_highgui3.so.3.1 => /opt/ros/kinetic/lib/libopencv_highgui3.so.3.1 (0x00007f2423828000)
    	libopencv_imgproc3.so.3.1 => /opt/ros/kinetic/lib/libopencv_imgproc3.so.3.1 (0x00007f2421e69000)
    	libopencv_core3.so.3.1 => /opt/ros/kinetic/lib/libopencv_core3.so.3.1 (0x00007f24211dc000)
    	libopencv_flann3.so.3.1 => /opt/ros/kinetic/lib/libopencv_flann3.so.3.1 (0x00007f241c29b000)
    	libopencv_imgcodecs3.so.3.1 => /opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1 (0x00007f241b7b8000)

And btw:

    pkg-config --list-all | grep cv
    opencv-3.1.0-dev                 OpenCV - Open Source Computer Vision Library
    opencv_apps                      opencv_apps - Description of opencv_apps
    cv_bridge                        cv_bridge - Description of cv_bridge

What's happening?

gauxonz, 2016-12-13:
Ok with two days of struggling here is my solution:
First my environment:
> Ubuntu 1604 under VMware
> ROS Kinetic
> OpenCV 2.4.10
I first tried to compile ORB_SLAM with the OpenCV3.1.0 inside ROS follow the introduction provided by  @cmxnono, however error occurs while cmake:
```
CMake Error in CMakeLists.txt:
  Imported target "opencv_xphoto" includes non-existent path

    "/usr/include/opencv-3.1.0-dev/opencv"

  in its INTERFACE_INCLUDE_DIRECTORIES.  Possible reasons include:

  * The path was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and references files it does not
  provide.
```
, as the make also failed unsurprisingly. I didn't spend lots of time on figure out why this happens (someone knows? ). Instead, I determine to install another OpenCV 3.1.0 aside of my original 2.4.10, into the path, say `\home\jack\libs\opencv`, then change findpackage for OpenCV in `CMakeList.txt` of ORB_SLAM to 
```
  find_package(OpenCV 3 REQUIRED HINTS ${OpenCV_DIR} NO_DEFAULT_PATH)
```
After compile the ORB_SLAM again, change the opencv version for ros node to 3, too, then everything goes well for me. 

By the way, I also tried to convert the rosbuild CMakeList file to catking_make version, it's doesn't help on this problem.

eugenelyj, 2018-01-05:
i met the same error with ROS indigo. Now i solved it by uninstalling opencv3 in my ubuntu14.04.
i search [ROS WIKI](http://wiki.ros.org) and find the sentence 

> opencv2 is the official version supported on Indigo and Jade

i used `ldd` to show shared libraries of `libORB_SLAM2.so` and `MonoAR` and i found that `libORB_SLAM2.so` is linked with opencv3 libraries, while `MonoAR` is linked with opencv2 and opencv3 libraries, showed as below (MonoAR):
```
	libopencv_imgproc.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_imgproc.so.2.4 (0x00007f67d053f000)
	libopencv_highgui.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_highgui.so.2.4 (0x00007f67d02f4000)
	libopencv_core.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_core.so.2.4 (0x00007f67cfebc000)
	libopencv_core.so.3.2 => /usr/local/lib/libopencv_core.so.3.2 (0x00007f67ced2b000)
	libopencv_imgproc.so.3.2 => /usr/local/lib/libopencv_imgproc.so.3.2 (0x00007f67cd4f7000)
	libopencv_contrib.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_contrib.so.2.4 (0x00007f67cac9b000)
	libopencv_calib3d.so.3.2 => /usr/local/lib/libopencv_calib3d.so.3.2 (0x00007f67c4db9000)
	libopencv_features2d.so.3.2 => /usr/local/lib/libopencv_features2d.so.3.2 (0x00007f67c4ae7000)
	libopencv_highgui.so.3.2 => /usr/local/lib/libopencv_highgui.so.3.2 (0x00007f67c48da000)
	libopencv_features2d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.2.4 (0x00007f67c24fd000)
	libopencv_calib3d.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.2.4 (0x00007f67c2266000)
	libopencv_ml.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_ml.so.2.4 (0x00007f67c1fee000)
	libopencv_video.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.2.4 (0x00007f67c1d9a000)
	libopencv_objdetect.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_objdetect.so.2.4 (0x00007f67c1b1f000)
	libopencv_flann.so.3.2 => /usr/local/lib/libopencv_flann.so.3.2 (0x00007f67ba0c1000)
	libopencv_imgcodecs.so.3.2 => /usr/local/lib/libopencv_imgcodecs.so.3.2 (0x00007f67b9e80000)
	libopencv_flann.so.2.4 => /usr/lib/x86_64-linux-gnu/libopencv_flann.so.2.4 (0x00007f67b980b000)
```
So that it causes some conflict, and i guess that it is ROS brought opencv2 in.

@gblack007  if you have sovled this problem, i think you should close it.

closed 2018-01-06 by gblack007

OAkyildiz, 2018-02-27:
I am almost pretty sure the image sizes you are declaring (or Mats you are initiating) and the images you are receiving from cv_bridge don't have the same size. If that is the case I suggest grabbing these variables from camera_info once before starting any image processing.

ns15417, 2019-10-29:
This problem is caused by the different version of Opencv. I solved this by rebuild the project. My problem was before: I run the code on Ubuntu16 and thus installed kinect and opencv3.* , but i didn't intend to use ros, so i build my project on opencv2.4. so later when i wanna to run ros again , it uses the opencv3.0 which is different from the one used to build the project.
How to solve it: Just rebuild the project based on opencv3.0 by add code below to CmakeLists.txt:
`find_package(OpenCV 3.0 QUIET)
if(NOT OpenCV_FOUND)
   find_package(OpenCV 2.4.3 QUIET)
   if(NOT OpenCV_FOUND)
      message(FATAL_ERROR "OpenCV > 2.4.3 not found.")
   endif()
endif()`

ihamieh, 2020-08-26:
@ns15417 which CmakeList.txt? The one inside ORB-SLAM2?? Please explain ... i've been having this problem for over two days!

CMing1994, 2020-11-01:
this is a opencv issue, please make sure that you use save version opencv, when you run ./build.sh and ./build_ros.sh. 
that means not only in the /ORB_SLAM2/CMakeLists.txt, 　you should use find_package( OpenCV 3.2.0  EXACT PATHS /home/OpenCV-3.2.0), but also in ORB_SLAM2/Examples/ROS/ORB_SLAM2/CMakeLists.txt  use
` find_package( OpenCV 3.2.0  EXACT PATHS /home/OpenCV-3.2.0)`   or the version you installed, and the certain path.
Besides, if it still not work,
you should check if the cv_bridge use a wrong version.
if so, 
`sudo apt-get remove ros_kinetic_cv_bridge`
and  
`cd catkin_ws/src 
git clone https://github.com/ros-perception/vision_opencv.git   
cd vision_opencv/cv_bridge`
in the CMakeLists :` find_package( OpenCV 3.2.0  EXACT PATHS /home/OpenCV-3.2.0)`
and then 
`catkin_make --pkg cv_bridge`
after this, 
./build.sh
and 
./buils_ros.sh

ihamieh, 2020-11-02:
thank @CMing1994 got that one resolved!