raulmur/ORB_SLAM2 #107: mono_tum.cc mono_kitti.cc undefined Reference to `ORB-SLAM2::System, vtable for Pangolin::Handler, HandlerScroll

ramtinkr, 2016-06-07:
Hi friends,

I have got an error while compiling mono_tum.cc and mono_kitti.cc. It would be so nice if you can help my in this way. 

[image]

Thanks
Ramtin

sinanariyurek, 2016-06-09:
To fix undefined reference from ORB_SLAM add PATH_TO_ORB_SRC/*.cc. (before -o)
Change -DECOMPILEWITHC11 to  -DCOMPILEWITHC11 
For pangolin add -LPATH_TO_PANGOLIN/build/src -lpangolin (after -o)

Moreover you will need to add paths of LIBS: g2o and DBoW2. Probably you will have to include paths of Eigen, Pangolin with -I, however I do not know your folder structure so this might not be a problem.

Note: After checking your previous post about ./build.sh not working your problem seems as it uses the wrong eigen library. CMAKELISTS display Eigen is included from /usr/include and you have an additional Eigen at you SLAM folder. Also std::chrono problem is solved by -DCOMPILEWITHC11 if your CMAKE does not add this simply add #define COMPILEWITHC11 after includes.

ramtinkr, 2016-06-09:
@sinanariyurek Thanks for your complete answer. I have tried another way to compile mono_tum.cc. I rebuilt it and used ./Examples/Monocular/mono_tum Vocabulary/ORBvoc.txt Examples/Monocular/TUMX.yaml PATH_TO_SEQUENCE_FOLDER and it was successfull. Now I have an error of rosbuild_init while compiling cmake .. in ROS path. By any chance, do you know how to solve it ? https://github.com/raulmur/ORB_SLAM2/issues/109

sinanariyurek, 2016-06-09:
I am not using ROS, so I do not not know why. Quick google http://answers.ros.org/question/37318/error-on-rosbuildcmake-cmake-error-at-cmakeliststxt/ maybe this might help.