raulmur/ORB_SLAM2 #108: fast motion loses track

antithing, 2016-06-08:
I have orbslam running with synchronized global shutter cameras, at 114 htz. This should be fast enough to mean there is no motion blur in the frames, but when I move the rig quickly, the track is lost. Is this related to motion blur? Is there a camera setting I can change to improve this? Or is there something I can tune in the code to improve the robustness at speed?

Thanks!

poine, 2016-06-08:
How are you grabbing your images? I doubt that you really are processing
frames at 115 Hz. It depends on the resolution, on the number of extracted
features and on your processor. For example, on my I7, with a resolution of
1280*1024 and 4000 features, i can process around 15fps

antithing, 2016-06-08:
Hi!  The processing is around 25fps,  I am grabbing the frames at 114 in another thread,  and pulling them in when orbslam is ready for the next.  Is there anything I can do to add stability for fast motion?

ank700, 2016-06-27:
Hi @poine , if the algorithm's image processing rate depends on the resolution, features and my processor, then does it make any sense to increase the camera fps? Or does camera fps help in increasing robustness to jerks or other sudden movements. 
In my experiments I observed that if I increase the camera fps to 70hz, the orb output rate increases by a few frames but if I increase it further to, say, 110 hz, then the orb rate decreases and sometimes I can see sudden jumps.(similar result with different number of maximum features)