raulmur/ORB_SLAM2 #113: on saving pose graph...

dannyway03, 2016-06-13:
Hi @raulmur.. and all..
I was just wondering how to store the complete pose graph related to a sequence. I guess it should be possible to store some *.g2o file from the optimization library, do you have any hint on how to do this in practice?
thanks,
-d

dannyway03, 2016-08-09:
Hi again @raulmur,
 
i still have problems while saving the pose graph, and i was wondering if i'm missing something obvious.
Currently, I'm calling optimizer.save(outputfilename.g2o) right after optimizer.optimize(nIterations) in Optimizer::BundleAdjustment(..): in this setup, i only get an empty file (although the optimizer contains plenty of vertices and edges).

While debugging this behavior, i found that actually no data is written be because the function factory->tag(v) return always an empty string (this function is within g2o in the thirdParty directory, it is called e.g., by OptimizableGraph::saveVertex(), file optimizable_graph.cpp line 820).
The very same applies to OptimizableGraph::saveEdge()..
Somehow, it seams that g2o does not recognize the graph of ORBSLAM.

Do you or anyone else have some hints on how to address this issue?
thanks,
-danny

mmattamala, 2017-04-17:
Hi, 

Despite this issue is a bit old, I was facing the same problem some days ago and I found the problem. Perhaps this can be useful for someone else:

The problem is that ORB-SLAM's modified g2o is not registering the g2o types, hence the Factory is not recognizing them as valid. For instance, here you have the `types_six_dof_expmap.cpp` in [Kuemmerle's g2o](https://github.com/RainerKuemmerle/g2o/blob/master/g2o/types/sba/types_six_dof_expmap.cpp), whereas here is [ORB-SLAM's](https://github.com/raulmur/ORB_SLAM2/blob/master/Thirdparty/g2o/g2o/types/types_six_dof_expmap.cpp). Please note the macros

    G2O_REGISTER_TYPE_GROUP(expmap);
    G2O_REGISTER_TYPE(VERTEX_SE3:EXPMAP, VertexSE3Expmap);
    G2O_REGISTER_TYPE(EDGE_SE3:EXPMAP, EdgeSE3Expmap);
    G2O_REGISTER_TYPE(EDGE_PROJECT_XYZ2UV:EXPMAP, EdgeProjectXYZ2UV);
    G2O_REGISTER_TYPE(EDGE_PROJECT_XYZ2UVU:EXPMAP, EdgeProjectXYZ2UVU);
    G2O_REGISTER_TYPE(PARAMS_CAMERAPARAMETERS, CameraParameters);

that are not included in ORB-SLAM; that's the trick. In addition, also note that Raúl modified this class to include new types such as `EdgeSE3ProjectXYZOnlyPose`, so you cannot visualize them with the original g2o viewer.

haithink, 2017-06-05:
Yes, mmattamala, you are right!